google-deepmind/mujoco #565: Parameter setting to prevent gradual slip of UR5 robot

zxwang96, 2022-11-07:
Hi,

I'm trying to control a simulated UR5 robot in mujoco. I found there is a gradual slipping behavior (seems driven by gravity) and tried to solve it by tune the parameter of `o_solimp` and `impratio ` as the official document suggested. But eventually I can't get a "stable" bot that its joints won't slip/move when there is no torque applied. Could you give me some tip of tuning these parameters? Thanks!

yuvaltassa, 2022-11-07:
you can try using the [noslip](https://mujoco.readthedocs.io/en/latest/modeling.html?highlight=noslip#algorithms-and-related-settings) solver. In general it is recommended to try all these things in an interactive setting (in `simulate`) so you can immediately see the effect of different parameters.

closed 2022-11-07 by yuvaltassa

zxwang96, 2022-11-07:
Thanks for the answer. My ur5 bot is manipulating a deformable object (rope), is it proper to use noslip solver in this situation?

yuvaltassa, 2022-11-07:
Well, it'll probably be a bit slow, since the noslip solver is based on PGS.
But if you don't care about that then no problem. 
I'm surprised impratio didn't solve the issue.

Play with all these things in `simulate` to understand the effects of different options.