google-deepmind/mujoco #571: The legacy mujoco forum cannot register an account

2371745431, 2022-11-09:
I want to download files on the legacy mujoco forum. Therefore, I need to sign up for an account. But I only see the login interface, and did not find the registration interface.

yuvaltassa, 2022-11-09:
File downloads should still work.

Which files specifically are you trying to download?

2371745431, 2022-11-15:
> 文件下载应该仍然有效。
> 
> 您具体要下载哪些文件？
Sorry, I still can't download the file. Click to download the file still prompted to login or register. Here is the link of the file I want to download:
https://roboti.us/forum/index.php?threads/6-bar-tensegrity-robot-model-optimization.3773/
https://roboti.us/forum/index.php?threads/modeling-of-highly-stiff-yet-elastic-structures.3822/#post-4988
https://roboti.us/forum/index.php?threads/nested-body-and-joint-type-for-tensegrity.3547/#post-4327

2371745431, 2022-11-15:
> File downloads should still work.
> 
> Which files specifically are you trying to download?
Is mujoco not support the modeling of parallel mechanisms? Suppose I want to create a triangle mechanism. The link should be named base_link-link1-link2 in sequence. How can I complete the closed chain and build the joint between link2 and base_link?

2371745431, 2022-11-15:
> File downloads should still work.
> 
> Which files specifically are you trying to download?

Sorry, I still can't download the file. Click to download the file still prompted to login or register. Here is the link of the file I want to download:
https://roboti.us/forum/index.php?threads/6-bar-tensegrity-robot-model-optimization.3773/
https://roboti.us/forum/index.php?threads/modeling-of-highly-stiff-yet-elastic-structures.3822/#post-4988
https://roboti.us/forum/index.php?threads/nested-body-and-joint-type-for-tensegrity.3547/#post-4327

yuvaltassa, 2022-11-16:
Try downloading the files now.

yuvaltassa, 2022-11-16:
Parallel mechanisms are supported. Look for example at the [Robotiq gripper](https://mujoco.readthedocs.io/en/latest/models.html#grippers-hands)

2371745431, 2022-11-19:
> 
Sorry, the file still cannot be downloaded. After clicking the file, the system prompts to register or log in.

2371745431, 2022-11-19:
> Try downloading the files now.

[image]

yuvaltassa, 2022-11-29:
Try again?
Should be okay now, I hope 🤞

yuvaltassa, 2022-11-30:
Links appear to work. Closing.

closed 2022-11-30 by yuvaltassa

2371745431, 2022-12-01:
> 

> Links appear to work. Closing.
Thanks for replying ! I can download files on the forum now. Thanks!