google-deepmind/mujoco #572: Cannot access all GPUs through EGL devices

Rocamonde, 2022-11-09:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a researcher and I'm trying to use MuJoCo for RL reward specification from natural language feedback.

I am trying to select the GPU device id that should be used by mujoco for rendering, so I can parallelize the rendering of some environment.

The documentation does not explicitly discuss this, so by reading through the Python/C code, I found that setting the EGL flag allows me to select this backend, which then when setting the MUJOCO_EGL_DEVICE_ID will select the corresponding GPU device.

However, currently only 1 device is shown as available when running `all_devices = EGL.eglQueryDevicesEXT()` as per the function `create_initialized_egl_device_display` in `mujoco.egl.__init__.py`. When  I run `nvidia-smi` I can see a list of 8 distinct GPU cards installed. This is running within a docker container using `nvidia/cuda:11.6.2-cudnn8-devel-ubuntu20.04`. 

Any tips on how to enable the rest of GPU devices?

saran-t, 2022-11-10:
I'm a Docker expert, but a colleague pointed to https://github.com/NVIDIA/nvidia-docker/wiki/. It might be worth you reading through Nvidia's documentation on how to configure GPU access inside Docker.

nimrod-gileadi, 2024-01-02:
Apologies if this is still an issue for you. Closing since there hasn't been any activity on this issue in a while.

closed 2024-01-02 by nimrod-gileadi

RealXuChe, 2024-10-17:
I encountered a similar issue. After investigating how to properly use Docker on a machine with NVIDIA GPUs, I’ve summarized the different problems I faced and my solutions here:

1. First, you should have the `nvidia-container-toolkit` installed on your host machine, and `nvidia-container-runtime` should be configured as Docker's backend runtime.
   You can check this by looking at the environment variables of the init process inside Docker: run `cat /proc/1/environ`. If the output contains something like `NVIDIA_DRIVER_CAPABILITIES=all NVARCH=x86_64 NVIDIA_REQUIRE_CUDA=cuda>=12.4 brand=tesla,driver>=470`, then this step has been successfully set up.

2. The `nvidia-container-toolkit` handles most of the tricky parts of passing the NVIDIA GPU into the container. However, by default, it doesn't enable graphical rendering functionality within the container. 
   If you encounter errors while trying to use GPU-accelerated rendering inside the container, you may need to create a new Docker container and pass the rendering capability into it when you do so. You can achieve this by passing the environment variable `-e NVIDIA_DRIVER_CAPABILITIES=all` when creating the container.

3. Using the environment variable above will add the driver libraries related to graphical rendering into the container. However, if the Docker image you are using doesn't include other components of the Linux graphics stack, you'll need to manually install them using a package manager. 
   For my Ubuntu image, I ended up needing to install the following three packages via apt-get: `libgl1`, `libx11-dev`, and `libglew-dev`. The missing graphics stack components in your image might differ from mine, so you should refer to the error messages you receive to install the necessary packages.

4. At this point, everything might already be working correctly. If it's not, and the error messages you see mention “MESA” or “nouveau,” it indicates that your graphics library is incorrectly calling the open-source graphics drivers. 
   Typically, your host machine will have proprietary graphics drivers installed, and the kernel module loaded will be the proprietary nvidia.ko. However, proprietary kernel drivers cannot work correctly with open-source user-space drivers, which leads to errors. 
   I found that the solution is to make your graphics libraries call the proprietary user-space graphics drivers. Specifically, the graphics library files on your system, such as libGLX or libEGL, are likely from the `libglvnd` project. These libraries don't contain actual graphical functionality but instead forward graphics API calls to the appropriate vendor libraries. `libglvnd` uses some json configuration files to determine which implementation to call. 
   In this case, the configuration file might not define the rule for dispatching to the NVIDIA proprietary implementation. If you're using EGL, you can check by running `ls /usr/share/glvnd/egl_vendor.d/`. For me, I could only see a `50_mesa.json file`. In this case, simply adding the following content to `/usr/share/glvnd/egl_vendor.d/10_nvidia.json` completely solved the problem for me:

```json
{
    "file_format_version" : "1.0.0",
    "ICD" : {
        "library_path" : "libEGL_nvidia.so.0"
    }
}
```

By applying all the steps above, I was able to use GPU rendering in Docker containers. I hope this helps you as well.

Boltzmachine, 2025-11-15:
Same error. but did not use docker