google-deepmind/mujoco #575: Reduce Euler's Disk effect?

Balint-H, 2022-11-10:
Hello quick question regarding modelling. This is something I've observed when both using the plugin, or simulate, on both mujoco 2.2.2 and 2.2.0 (running on Windows).

I'm roughly modelling a coffee cup with a few primitives, with a cylinder at the bottom. I've noticed that it's very easy for the cup to go into a [Euler's Disk](https://en.wikipedia.org/wiki/Euler%27s_Disk) like state, rolling on the cylinders bottom edge. I've originally built the scene with the Unity plugin, and exported  the XML, which I've attached below. I've removed unnecessary components and confirmed the effect is still there.

I'm not too surprised that this effect happens, but was wondering whether there's any advice that I could use to configure the geoms in a way to discourage the spinning, or make it converge faster? I tried increasing the rolling and sliding friction, or play around with object density, with no success. 

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
  <worldbody>
    <body pos="0 0 0.0644" quat="-1 0 0 0" name="mj_coffee_0">
      <geom density="1000" type="cylinder" size="0.06 0.025" pos="0 0 0.0298" quat="-1 0 0 0" priority="0" contype="1" conaffinity="1" group="0" condim="3" solmix="1" solref="0.02 1" solimp="0.9 0.95 0.001" margin="0" gap="0" friction="1 0.005 0.0001" name="coffee_geom1_1" />
      <geom density="1000" type="box" size="0.02 0.005 0.04" pos="-0.05189998 0 0.002899997" quat="-0.902435 0 0.4308261 0" priority="0" contype="1" conaffinity="1" group="0" condim="3" solmix="1" solref="0.02 1" solimp="0.9 0.95 0.001" margin="0" gap="0" friction="1 0.005 0.0001" name="handle_geom_2" />
      <freejoint name="freejoint_3" />
      <geom density="2500" type="cylinder" size="0.04 0.025" pos="0 0 -0.0274" quat="-1 0 0 0" priority="1" contype="1" conaffinity="1" group="0" condim="3" solmix="1" solref="0.02 1" solimp="0.9 0.95 0.001" margin="0" gap="0" friction="2 0.005 1" name="coffee_geom2_4" />
    </body>
  <geom density="1000" type="plane" size="0.5 0.5 1" pos="0 0 0" quat="-1 0 0 0" priority="0" contype="1" conaffinity="1" group="0" condim="3" solmix="1" solref="0.02 1" solimp="0.998 0.999 0.0001" margin="0" gap="0" friction="1 0.005 0.0001" name="geom_0_18" />
  </worldbody>
</mujoco>
```

</details>

Here is a screenshot / video, illustrating my question:

yuvaltassa, 2022-11-10:
This is because box-cylinder collisions only generate 1 contact point. You currently have two options:

1. Use a plane instead of a box. Plane-cylinder collisions have 3 points rather than just 1.
2. Enable the `multiccd` option. This will also generate more contact points when the cylinder is near flat.

closed 2022-11-10 by yuvaltassa

Balint-H, 2022-11-10:
Doesn't the scene already use a cylinder and plane contact? The box geom is only there to model the handle of the coffee mug.

yuvaltassa, 2022-11-10:
Can you visualise contact points and let me know if there are 3 or 1?

Balint-H, 2022-11-10:
There are 3 points while its steady on the plane, but as soon as it is on its edge, only one is visible (or maybe they just coincide?). At this point if it gets the right amount of roll, the oscillations start, and can keep going for over a minute. I applied some downward force to end this video early:

https://user-images.githubusercontent.com/41113387/201102865-01d8a3f6-294a-46e1-b314-173d74ba280f.mp4

yuvaltassa, 2022-11-10:
Oh I see, that's what you mean.

This is not a bug, this is a feature. Your system is energy conserving, spinning forever is exactly what it's supposed to do. You are used to living in a world with air and microscopic, energy-depleting deformations at the point of contact. Neither of these things are true by default in MuJoCo, so energy is conserved. You can increase air density and viscosity but most importantly, increase the [dimensionality of you contacts](https://mujoco.readthedocs.io/en/latest/computation.html#contact). And possibly increase rolling and torsional friction.

Give your cylinder `<geom condim="6"/>` and set the appropriately. The [default values](https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-geom) for torsional and rolling friction are respectively `5e-3` and `1e-4`, have units of length, and should be interpreted as the scale of the material being deformed at the contact point. I should update the docs to explain this actually.

Balint-H, 2022-11-23:
Only got around to test this just now. Just to confirm in case someone comes here in the future, it was bumping up `condim` to 6 that did the trick. I had not looked into it before, and did not understand why the rolling friction was not applied. Thanks again!