google-deepmind/mujoco #577: How can I simulate a breakable solid or a liquid?

chanind, 2022-11-10:
Hi,

I'm a student and I'm trying to use MuJoCo for simulating physics for common-sense knowledge relating to physical reasoning.

I'm looking for some help working with liquids and breakable solids. For instance, is it possible to simulate a glass of water falling off a table? I would expect the glass to shatter and the water to spill on the ground in the final state. I'm struggling to figure out how to specify this in MJCF. I see there are viscosity and density options in the XML which seem like they might relate to liquids, but I can't figure out how to specify that a body should be treated as a liquid. I also can't find any reference to anything related to brittleness or the ability to break on impact.

Apologies if I'm missing something obvious!

quagla, 2022-11-10:
Hi, thanks for the message. There is currently no support for fluid simulations in MuJoCo, but it's something I'd be keen to add. However, it's unlikely that it will happen in the near future. That's quite a sophisticated (and interesting) scenario that you want to simulate. I'm sorry I can't give more help but I don't see a quick solution to achieve this at the moment.

yuvaltassa, 2022-11-10:
You can simulate breaking by attaching multiple bodies with weld constraints and disabling the constraints when the force or the displacement is greater than X.

Discussed in #197 and #525.

closed 2022-11-10 by yuvaltassa