google-deepmind/mujoco #578: mujoco on GPU support

MyelinsheathXD, 2022-11-10:
Hello! 
After observing some of the samples of fully GPU training of [OmniIsaccGymEnv](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs), I noticed that the fully GPU traning system is the way to achive fastest traning method for any small physical interactions.
Typically, small Envirounments like Ant, Humanoid walking   can run  up to 4000 instancess in a single GPU in real -time.

Traning time also a lot more faster.

Any plans to support Mujoco on GPU? .

closed 2022-11-10 by MyelinsheathXD

BaiYunpeng1949, 2023-01-20:
Hi, 

May I ask why you closed this thread as completed? Did you find out any solutions?

Thank you.

MyelinsheathXD, 2023-01-21:
> Hi,
> 
> May I ask why you closed this thread as completed? Did you find out any solutions?
> 
> Thank you.

Hey! 
After seeing  this closed issue
https://github.com/deepmind/mujoco/issues/85 , I noticed that the team is not actively involved to implament GPU version of MUJOCO.

My solution is just stick with Phyx 5.1 and Isacc gym. Since even for Nvidia team , it's too hard to properly implament GPU Physics version for low spec GPUs. only High end GPUs that have large VRAM can properly train on only inside a  GPU.

  
It might take too much time to implement mujoco-GPU to all  GPU vendors for OpenAI team.  And this requires high-end gpu all the time to simulate large environments and train them in a single GPU

BaiYunpeng1949, 2023-01-21:
> Hey! After seeing this closed issue #85 , I noticed that the team is not actively involved to implament GPU version of MUJOCO.
>
> My solution is just stick with Phyx 5.1 and Isacc gym. Since even for [...]

Got it, thank you for the elaborated reply! I am also looking for a solution to boost up my RL + MuJoCo. Will try Isacc gym as well.