google-deepmind/mujoco #581: Build a human like agent in mujoco via skin or mesh

friedmainfunction, 2022-11-14:
Hi, I'm new in mujoco community, it is so cool! 
And I'm trying to use MuJoCo to build a human like body for simulation, just like this paper does in **Isaac Gym**:
[image]
[image]
https://xbpeng.github.io/projects/ASE/index.html
I noticed that they use a humanoid body modeled in xml format, and then add a wonderful visualization, but it still simulates in simple humanoid in essential which corresponds to skin mechanism in mujoco. So, I am trying to do the same thing in mujoco.

Basically, I am trying to use the simple humanoid xml files as same as paper above and additional 3D skins with textures to do it.
I have tried to utilize your example in https://github.com/deepmind/dm_control/blob/main/dm_control/suite/dog.xml, but details of it are omitted. 
**Here are the questions which I am faced with**:
1. I only have .obj mesh files, and I don't know how to transform it into .skn format automatically. I think it's hard to describe all vertexes manually and I don't really understand how to bind the skins (suppose that I have them) with humanoid bones.
2. Should I do it via skins? If you find I am in a wrong direction, please tell me the better solution!

I am working with Python bindings of mujoco-2.3.0 and Ubuntu 20.04.
Here is the humanoid xml file: https://github.com/nv-tlabs/ASE/blob/main/ase/data/assets/mjcf/amp_humanoid.xml

It is appreciated if you can give any advice.

yuvaltassa, 2022-11-14:
Yes, you should use skins.

Check out #128 for details.

closed 2022-11-14 by yuvaltassa

friedmainfunction, 2022-11-17:
> Yes, you should use skins.
> 
> Check out #128 for details.
@yuvaltassa 
Thank you for replying!
Before I open this issue, I have checkout that.
What I still do not understand is how to determine the  **bind_postion**, **vertex_index** and **vertex_weight**. I write a python script to show the data in your dog_skin.skn file, but I can't find the law in these data, and I don't know how to determine thousands positions and weights. Are there any tools or methodology to solve this problem? Please give me some advice, thanks.
The data in your dog_skin.skn file is just like that:
[image]
which I don't know how to generate.

yuvaltassa, 2022-11-18:
[here](https://github.com/deepmind/dm_control/blob/main/dm_control/locomotion/walkers/assets/dog_v2/build_dog.py) is an example, it's the script that generates the vertex weights for the dog. (under "make skin")