google-deepmind/mujoco #583: How to simulate a camera at a fixed position that can send compressed frames on demand ?

joaopfg, 2022-11-15:
Hi, in my simulation I need to add some cameras at fixed positions to the scene and request frames from it on demand (let's say, when clicking a button in the UI for now), which will be compressed and sent to an external service via some communication protocol.

So far, I could learn the following from the docs:

1) I can define a cameras in the world body in the MJCF like the following:

```
<worldbody>
    <camera name="camera1" pos="..." quat="..."/>
    <camera name="camera2" pos="..." quat="..."/>
    ...
<worldbody/>
```

2) I can access access the following camera info from `mjModel`:

```
  int*      cam_mode;             // camera tracking mode (mjtCamLight)       (ncam x 1)
  int*      cam_bodyid;           // id of camera's body                      (ncam x 1)
  int*      cam_targetbodyid;     // id of targeted body; -1: none            (ncam x 1)
  mjtNum*   cam_pos;              // position rel. to body frame              (ncam x 3)
  mjtNum*   cam_quat;             // orientation rel. to body frame           (ncam x 4)
  mjtNum*   cam_poscom0;          // global position rel. to sub-com in qpos0 (ncam x 3)
  mjtNum*   cam_pos0;             // global position rel. to body in qpos0    (ncam x 3)
  mjtNum*   cam_mat0;             // global orientation in qpos0              (ncam x 9)
  mjtNum*   cam_fovy;             // y-field of view (deg)                    (ncam x 1)
  mjtNum*   cam_ipd;              // inter-pupilary distance                  (ncam x 1)
  mjtNum*   cam_user;             // user data                                (ncam x nuser_cam)
```

3) I can access the following info from `mjData`:

```
mjtNum*   cam_xpos;             // Cartesian camera position                (ncam x 3)
mjtNum*   cam_xmat;             // Cartesian camera orientation             (ncam x 9)
```

But I can't finish my task with this information. So, I would like to ask:

1) How to set `ncam` ?

2) How to request frames from one of the cameras ?

3) Could you provide a picture of the way you model a camera in MuJoCo, please ? So far, what I have in my mind is something like in the following image:
[image]

Is it it close to the MuJoCo model ? If so, how to determine the distance `f` with the data MuJoCo gives ? And what about the projection surface position and orientation ? How to set it ?

joaopfg, 2022-11-16:
UPDATE:

I made some progress. I call the following piece of code when clicking on a button, which allows me to get an rgb shot from a fixed camera defined in my MJCF file:

```
            this->cam.type = mjCAMERA_FIXED;
            this->cam.fixedcamid = 0;
            this->camera = 2 + this->cam.fixedcamid;
            const unsigned int h = this->uistate.rect[3].height;
            const unsigned int w = this->uistate.rect[3].width;
            std::unique_ptr<unsigned char[]> rgb(new unsigned char[3*w*h]);
            if (!rgb) {
                mju_error("could not allocate buffer for screenshot");
            }
            mjr_readPixels(rgb.get(), NULL, this->uistate.rect[3], &this->con);
```

However, by doing this the central simulation screen also changes, which I don't want. I am looking at the `record.cc` example, which sets an offscreen buffer in a global `mjrContext` object by calling `mjr_setBuffer(mjFB_OFFSCREEN, &con)`. This seems exactly like what I want.  

How should I change `simulate.cc` to also associate offscreen buffers to each fixed camera of my model (while keeping the visualization from the free camera in the central screen) ?

DavidPL1, 2022-11-19:
It's unclear to me what `this->camera` is.

Otherwise your code snipped is actually missing a call to `mjv_updateScene` with the updated camera id and afterwards a call to `mjr_render` to update the context buffer accordingly, before copying the content to your rgb buffer.

You could try the following which restores the camera id and type after writing to the rgb buffer:
```c
mjtCamera type_backup = this->cam.type;
int id_backup = this->cam.fixedcamid;

this->cam.type = mjCAMERA_FIXED;
this->cam.fixedcamid = 2; // id of the camera you want to render
const unsigned int h = this->uistate.rect[3].height;
const unsigned int w = this->uistate.rect[3].width;
std::unique_ptr<unsigned char[]> rgb(new unsigned char[3*w*h]);
if (!rgb) {
    mju_error("could not allocate buffer for screenshot");
}

mjv_updateScene(this->model, this->data, &this->opt, NULL, &this->cam, mjCAT_ALL, this->scn); // Change variable names accordingly
mjr_render(this->uistate.rect[3], &this->scn, &this->con);
mjr_readPixels(rgb.get(), NULL, this->uistate.rect[3], &this->con);

this->cam.type = type_backup;
this->cam.fixedcamid = id_backup;
```
This would reuse all resources used by the onscreen renderer. You could also use a separate invisible glfw window, camera, scene, and context that not interfere with the onscreen rendering, to render off-screen.

joaopfg, 2022-11-20:
@DavidPL1 Thanks a lot ! Could you show me the code to render off-screen, please ?

closed 2022-12-07 by kevinzakka