google-deepmind/mujoco #586: Collapsible subsections in right TOC sidebar in Docs

vaxenburg, 2022-11-16:
**Is your feature request related to a problem? Please describe.**

The right TOC sidebar in the new docs theme is far too long and not useful, especially in long reference sections such as [API Reference](https://mujoco.readthedocs.io/en/latest/APIreference.html#). A lot of scrolling is required to reach the destination and it's also hard to find in the long list now. It makes using the TOC to navigate efficiently through the long reference sections almost impossible.

**Describe the solution you'd like**

Please consider making sublevel elements collapsible, and collapsed by default. The previous docs theme worked perfectly fine -- upon opening everything was collapsed and the TOC showed only the first level elements. It was very easy to navigate. In the same API Reference section, the first level of elements looked like this, and you could reach your desired section in a couple of clicks:

[image]

kevinzakka, 2022-11-19:
Looking into this, thanks for flagging @vaxenburg!

vaxenburg, 2023-01-18:
Any update on this, by any chance?

kevinzakka, 2023-02-21:
Hi @vaxenburg, the docs are slowly being broken up to alleviate this issue (e.g., 090c25988ae2d4df2525ab9ef93658824ffe6b86).

yuvaltassa, 2023-02-22:
Closing for now.

vaxenburg, 2023-02-22:
Sounds good, thanks @kevinzakka !

closed 2023-04-28 by kevinzakka