google-deepmind/mujoco #595: About Semantic Segmentation

Lucky-hqj, 2022-11-19:
Hi, I'm a novice using mujoco. How to use Python Bindings to obtain semantic map?

yuvaltassa, 2022-11-19:
What is a semantic map?

Lucky-hqj, 2022-11-19:
> What is a semantic map?

Sorry, there is something wrong with my statement. I mean, image semantic segmentation.

yuvaltassa, 2022-11-19:
We will shortly (likely next week) make a colab available with an example of this functionality. 
I will leave this issue open until this happens.
Cheers

Lucky-hqj, 2022-11-20:
> We will shortly (likely next week) make a colab available with an example of this functionality. I will leave this issue open until this happens. Cheers

OK.I'm looking forward to it.Thank you very much for your contribution.

kbayes, 2022-11-29:
The colab is now online (located in the Python subdir, or find the link in the README).  See the advanced graphics section for segmentation.  Let us know if anything is unclear or missing, so we can improve it for other users.

closed 2022-12-06 by kevinzakka