google-deepmind/mujoco #596: Setting segids for segmentation rendering

DavidPL1, 2022-11-19:
Hi,

I'm currently working on segmentation image rendering using the C++ API and I was wondering how segids should be set.
While this can easily be done manually in code by accessing the geoms and setting their segid property, this requires model dependent code, hence I would like to set the ids directly in the xml.
I couldn't find anything in the documentation or the code, did I miss something or is this currently not possible? And if it's not, is this already planned as a feature?

yuvaltassa, 2022-11-20:
This question is confusing. You don't get to set segid, the renderer does that for you.

segid's correspond to the index of the visual geom (so includes sites, frames joints etc.). You probably want to convert those id's back to `mjModel` ids. [Here is some Python code](https://github.com/deepmind/dm_control/blob/e3d4c0aecfd710cd7428eb77af203dc5fbdacc0f/dm_control/mujoco/engine.py#L886) that does this.

closed 2022-11-20 by yuvaltassa

DavidPL1, 2022-11-21:
Oh, then I think I misunderstood the purpose of segids and the use of the mjRND_IDCOLOR flag. And I was a little unclear with how I phrased my question.
 
My intuition was that mjRND_IDCOLOR set to false would get you random colors (left image) and setting it to true would let me somehow be in control of which color is assigned to which geom. Instead I get the image to the right, rendering different shades of red.

[image] [image]

To clearly explain what I would like to do:
I would like to set the colors in the segmented image directly in the model file. For instance, all links of the robot in the same color, the gripper of the robot in another color and so on.

The use case for this is as follows: I am able to configure that for a camera in a certain interval both a normal RGB image and an RGB image with the chosen segmentation colors are saved. Once these image pairs are externally processed, for each normal RGB image I would have an annotated segmentation mask of user-defined object-classes consisting of an arbitrary number of geoms.

Is anything of the likes planned or do you consider this too much of an edge-case for a feature?

yuvaltassa, 2022-11-21:
This feature is not planned. `mjRND_IDCOLOR` means "random colours", you don't get to set them.

However, using the regular visual geom indexing (which look like dark red when rendered, due to the integer->RGB conversion), , as explained above, you can do this yourself pretty easily.

juanmed, 2022-11-24:
@yuvaltassa  Hi, thanks for the details. I checked the [python example](https://github.com/deepmind/dm_control/blob/e3d4c0aecfd710cd7428eb77af203dc5fbdacc0f/dm_control/mujoco/engine.py#L886) you linked above. Do you have an example of the same lines to extract the 2D mask image (objid, objtype) but using the C++ interface?  Thanks.

yuvaltassa, 2022-11-28:
>When setting mjRND_SEGMENT each channel of the pixel values obtained from mjr_readPixels are simply random numbers in (0,255) range, right? i.e. (there is not objid, objtype information encoded)

Correct.

> When setting mjRND_IDCOLOR, what is the meaning of each of the 3 channels of each pixel returned by mjr_readPixels?

See below.

> Does it make sense to set both mjRND_SEGMENT and mjRND_IDCOLOR? it seems this renders the same result as if using setting only mjRND_IDCOLOR?

One is a modifier of the other. mjRND_SEGMENT gives you the random colors. mjRND_IDCOLOR means use colours that have id semantics, see below. This is documented [here](https://mujoco.readthedocs.io/en/latest/APIreference.html?highlight=mjRND_IDCOLOR#mjtrndflag)

> Why are the channels of self._rgb_buffer converted to int32 and the summed? Does this mean that all the channels together encode the objid?

Yes. The semantic of mjRND_IDCOLOR is **index into the geoms buffer of the scene** -- [this array](https://github.com/deepmind/mujoco/blob/fdb81f3388f9d04b6f9dea4ca4a19e07259301bf/include/mujoco/mjvisualize.h#L274). Note that these are **visual geoms**, i.e. including decorations like frame visualisations. You probably want to go into the visual geom itself to read its [type and id](https://github.com/deepmind/mujoco/blob/fdb81f3388f9d04b6f9dea4ca4a19e07259301bf/include/mujoco/mjvisualize.h#L206) in the `mjModel`. This is what the Python code is doing.

juanmed, 2022-11-30:
Thanks for the detailed answer. I saw mjRND_SEGMENT and mjRND_IDCOLOR in the docs, I guess I did not catch the exact meaning at first. Just to confirm I understand this correct:

1) To get id semantics in the 2D mask image I must set mjRND_SEGMENT *and* mjRND_IDCOLOR (by experimentation this seems to be the case)
2) Recover all unique int32 from the mask image (as in the python example). These correspond to the `mjvGeom.segid` of the visible geoms.
3) (I am a bit unclear on this one) The recovered `mjvGeom.segid`s, are these the index of the corresponding geom in `mjvScene.geoms`?  This touches upon my next point.
4) I do want to get the type and id of the visual geom itself as you well mentioned, and at some point access other values such as its position, orientation, etc. To access the visual geom itself,  I am attempting:

```
# I receive one "segid" as input from the mask 

int objid;
int objtype;
int geom_index;
for ( i = 0; i < scn->ngeoms; ++i) {
    if ( scn->geoms[i].segid == segid) {
        objid = scn->geoms[i].objid;
        objype = scn->geoms[i].objtype;
        geom_index = i;
    }
}

```
This is what I understood from the python example. Would this recover the correct `objid` and `objtype`?

5) How do I index other fields related to the geom corresponding to the `segid` I am interested? Should I use `objid`, `geom_index`. I have tried, for example:

```
To get the global position
mjData* d;
position << d->geom_xpos[3 * objid + 0], d->geom_xpos[3 * objid + 1], d->geom_xpos[3 * objid + 2];
orientation << d->geom_xpos[9 * objid + 0], d->geom_xpos[9 * objid + 1] ........ d->geom_xpos[3 * objid + 0], d->geom_xpos[9 * objid + 8];
```
Is this the correct way to index the global position and orientation of the geom corresponding to a segid?

Thanks for your time. I have already searched in the documentation and tried several things by myself, but the positions/orientations I get look wrong.

Thanks

juanmed, 2022-12-01:
To answer one of my questions, at least for the geoms in mjvScene, it seems that its `segid` and its index `i` in `mjvScene.geoms` are the same value.