google-deepmind/mujoco #605: Simulate :  Windows Makefile error : LNK2001, LNK2019: unresolved external symbols

rohit-kumar-j, 2022-11-24:
- Operating system: Windows 10 (x64 Native Tools 2022)
- MuJoCo version Release : 2.3.0  ([mujoco-2.3.0-windows-x86_64.zip](https://github.com/deepmind/mujoco/releases/download/2.3.0/mujoco-2.3.0-windows-x86_64.zip))

Running make in `simulate/` causes the following

```
********************************************************************** 
** Visual Studio 2022 Developer Command Prompt v17.1.0 
** Copyright (c) 2021 Microsoft Corporation                                                                            
**********************************************************************                                                  
[vcvarsall.bat] Environment initialized for: 'x64' 

$ C:\Users\User\Mujoco\simulate› make 
                                                                          
cl /O2 /MT /EHsc /arch:AVX /I../include /Fe../bin/ main.cc simulate.cc uitools.cc ../lib/glfw3dll.lib ../lib/mujoco.lib Microsoft (R) C/C++ Optimizing Compiler Version 19.31.31104 for x64 
                                                    
Copyright (C) Microsoft Corporation.  All rights reserved.                                                                                                                                                                                      main.cc                                                                                                                 
C:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\shared\minwindef.h(130): warning C4005: 'APIENTRY': macro redefinition                                                                                                            
../include\GLFW/glfw3.h(123): note: see previous definition of 'APIENTRY'                                               
simulate.cc                                                                                                             
simulate.cc(1700): warning C4273: 'mujoco::Simulate::applyposepertubations': inconsistent dll linkage                   C:\Users\User\Mujoco\simulate\simulate.h(52): note: see previous definition of 'applyposepertubations'                 
simulate.cc(1709): warning C4273: 'mujoco::Simulate::applyforceperturbations': inconsistent dll linkage                 C:\Users\User\Mujoco\simulate\simulate.h(55): note: see previous definition of 'applyforceperturbations'               
simulate.cc(1718): warning C4273: 'mujoco::Simulate::load': inconsistent dll linkage                                    C:\Users\User\Mujoco\simulate\simulate.h(59): note: see previous definition of 'load'                                  
simulate.cc(1741): warning C4273: 'mujoco::Simulate::loadmodel': inconsistent dll linkage                               C:\Users\User\Mujoco\simulate\simulate.h(63): note: see previous definition of 'loadmodel'                             
simulate.cc(1821): warning C4273: 'mujoco::Simulate::prepare': inconsistent dll linkage                                 C:\Users\User\Mujoco\simulate\simulate.h(66): note: see previous definition of 'prepare'                               
simulate.cc(1883): warning C4273: 'mujoco::Simulate::render': inconsistent dll linkage                                  C:\Users\User\Mujoco\simulate\simulate.h(69): note: see previous definition of 'render'                                
simulate.cc(2056): warning C4273: 'mujoco::Simulate::clearcallback': inconsistent dll linkage                           C:\Users\User\Mujoco\simulate\simulate.h(72): note: see previous definition of 'clearcallback'                         
simulate.cc(2061): warning C4273: 'mujoco::Simulate::renderloop': inconsistent dll linkage                              C:\Users\User\Mujoco\simulate\simulate.h(76): note: see previous definition of 'renderloop'                            
uitools.cc                                                                                                              
Generating Code...                                                                                                      
Microsoft (R) Incremental Linker Version 14.31.31104.0                                                                  
Copyright (C) Microsoft Corporation.  All rights reserved.                                                                                                                                                                                      
/out:../bin/main.exe                                                                                                    
main.obj                                                                                                                
simulate.obj                                                                                                           
 uitools.obj                                                                                                             
../lib/glfw3dll.lib                                                                                                     
../lib/mujoco.lib                                                                                                          
Creating library ..\bin\main.lib and object ..\bin\main.exp                                                          
LINK : warning LNK4217: symbol '?applyposepertubations@Simulate@mujoco@@QEAAXH@Z (public: void __cdecl mujoco::Simulate::applyposepertubations(int))' defined in 'simulate.obj' is imported by 'main.obj' in function '"void __cdecl `anonymous namespace'::PhysicsLoop(class mujoco::Simulate &)" (?PhysicsLoop@?A0x203b6543@@YAXAEAVSimulate@mujoco@@@Z)'             LINK : warning LNK4217: symbol '?applyforceperturbations@Simulate@mujoco@@QEAAXXZ (public: void __cdecl mujoco::Simulate::applyforceperturbations(void))' defined in 'simulate.obj' is imported by 'main.obj' in function '"void __cdecl `anonymous namespace'::PhysicsLoop(class mujoco::Simulate &)" (?PhysicsLoop@?A0x203b6543@@YAXAEAVSimulate@mujoco@@@Z)'         
LINK : warning LNK4217: symbol '?load@Simulate@mujoco@@QEAAXPEBDPEAUmjModel_@@PEAUmjData_@@_N@Z (public: void __cdecl mujoco::Simulate::load(char const *,struct mjModel_ *,struct mjData_ *,bool))' defined in 'simulate.obj' is imported by 'main.obj' in function '"void __cdecl `anonymous namespace'::PhysicsLoop(class mujoco::Simulate &)" (?PhysicsLoop@?A0x203b6543@@YAXAEAVSimulate@mujoco@@@Z)'                                                                                      
LINK : warning LNK4217: symbol '?renderloop@Simulate@mujoco@@QEAAXXZ (public: void __cdecl mujoco::Simulate::renderloop(void))' defined in 'simulate.obj' is imported by 'main.obj' in function 'main'                                          main.obj : error LNK2019: unresolved external symbol "struct mujoco::Glfw const & __cdecl mujoco::Glfw(void *)" (?Glfw@mujoco@@YAAEBU01@PEAX@Z) referenced in function "void __cdecl `anonymous namespace'::PhysicsLoop(class mujoco::Simulate &)" (?PhysicsLoop@?A0x203b6543@@YAXAEAVSimulate@mujoco@@@Z)                                                              simulate.obj : error LNK2001: unresolved external symbol "struct mujoco::Glfw const & __cdecl mujoco::Glfw(void *)" (?Glfw@mujoco@@YAAEBU01@PEAX@Z)                                                                                             
uitools.obj : error LNK2001: unresolved external symbol "struct mujoco::Glfw const & __cdecl mujoco::Glfw(void *)" (?Glfw@mujoco@@YAAEBU01@PEAX@Z)                                                                                             
main.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl mujoco::Simulate::Simulate(void)" (__imp_??0Simulate@mujoco@@QEAA@XZ) referenced in function "class std::unique_ptr<class mujoco::Simulate,struct std::default_delete<class mujoco::Simulate> > __cdecl std::make_unique<class mujoco::Simulate,0>(void)" (??$make_unique@VSimulate@mujoco@@$$V$0A@@std@@YA?AV?$unique_ptr@VSimulate@mujoco@@U?$default_delete@VSimulate@mujoco@@@std@@@0@XZ)       
main.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl mujoco::Simulate::~Simulate(void)" (__imp_??1Simulate@mujoco@@QEAA@XZ) referenced in function "public: __cdecl std::unique_ptr<class mujoco::Simulate,struct std::default_delete<class mujoco::Simulate> >::~unique_ptr<class mujoco::Simulate,struct std::default_delete<class mujoco::Simulate> >(void)" (??1?$unique_ptr@VSimulate@mujoco@@U?$default_delete@VSimulate@mujoco@@@std@@@st [... cut ...] nsigned int)" (?encode@lodepng@@YAIAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@PEBEIIW4LodePNGColorType@@I@Z) referenced in function "public: void __cdecl mujoco::Simulate::render(void)" (?render@Simulate@mujoco@@QEAAXXZ)                                                                                                ..\bin\main.exe : fatal error LNK1120: 5 unresolved externals                                                           
make: *** [Makefile:20: all] Error 2 
```

rohit-kumar-j, 2022-11-24:
The makefile
```
#WINDOWS
COMMON=/O2 /MT /EHsc /arch:AVX /I../include /Fe../bin/
all:
	cl $(COMMON) main.cc simulate.cc uitools.cc ../lib/glfw3dll.lib ../lib/mujoco.lib
	del *.obj
```

mhming, 2022-12-23:
[image]
I get the same error in 2.3.0 & 2.3.1, how to solve?

yuvaltassa, 2022-12-23:
Hi both,

Sorry for late response.

As mentioned elsewhere (search issues for "makefile"), we are not really keeping the makefiles up to date and are leaning towards deleting them completely in favour of cmake only.

Do you have a good reason in favour of keeping the makefiles?

mhming, 2022-12-29:
I think cmake only is a good choice.
However, lodepng.png & CmakeLists is missing from your published documents, I have written a CmakeLists file for windows, linux and macos for reference.

cmake_minimum_required(VERSION 3.22)
project(shsEnv)

set(CMAKE_CXX_STANDARD 17)

set(CMAKE_SHARED_LINKER_FLAGS "${CMAKE_CXX_FLAGS} -Wl,-no-as-needed")

add_definitions(-DMJSIMULATE_STATIC) # use static =,solve the "error LNK2001/LNK2019" in windows

# set MUJOCO root
set(MUJOCO_HOME /opt/mujoco)
include_directories(${MUJOCO_HOME}/include/)
link_directories(${MUJOCO_HOME}/lib/)

set(COM_SRC
        mujoco/glfw_dispatch.cc
        mujoco/lodepng.cpp
        mujoco/simulate.cc
        mujoco/uitools.cc
        mujoco/main.cc)

set(COM_LIB mujoco glfw)

if(APPLE)
set(COM_SRC ${COM_SRC} mujoco/macos_gui.mm)
set(COM_LIB ${COM_LIB} "-framework Cocoa")
endif()

add_executable(${PROJECT_NAME} ${COM_SRC})

target_link_libraries(${PROJECT_NAME} ${COM_LIB})

yuvaltassa, 2024-01-08:
I believe this is now resolved?

Closing for now.

closed 2024-01-08 by yuvaltassa