google-deepmind/mujoco #607: Build issues with python 3.11

pseudo-rnd-thoughts, 2022-11-24:
In the python `setup.py` file, the python 3.11 classifier exists however in pypi, a python 3.11 precompiled version is missing. 
Therefore, when I run `pip install mujoco`, I get the following error 

```shell
(Gymnasium3.11) marktowers@UOS-18898 phd % pip install mujoco
Collecting mujoco
  Using cached mujoco-2.3.0.tar.gz (144 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: absl-py in /Users/marktowers/opt/anaconda3/envs/Gymnasium3.11/lib/python3.11/site-packages (from mujoco) (1.3.0)
Requirement already satisfied: glfw in /Users/marktowers/opt/anaconda3/envs/Gymnasium3.11/lib/python3.11/site-packages (from mujoco) (2.5.5)
Requirement already satisfied: numpy in /Users/marktowers/opt/anaconda3/envs/Gymnasium3.11/lib/python3.11/site-packages (from mujoco) (1.23.5)
Requirement already satisfied: pyopengl in /Users/marktowers/opt/anaconda3/envs/Gymnasium3.11/lib/python3.11/site-packages (from mujoco) (3.1.6)
Building wheels for collected packages: mujoco
  Building wheel for mujoco (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [60 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.macosx-10.9-x86_64-cpython-311
      creating build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/rollout.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/rollout_test.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/bindings_test.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/render_test.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      creating build/lib.macosx-10.9-x86_64-cpython-311/mujoco/egl
      copying mujoco/egl/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/egl
      copying mujoco/egl/egl_ext.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/egl
      creating build/lib.macosx-10.9-x86_64-cpython-311/mujoco/osmesa
      copying mujoco/osmesa/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/osmesa
      creating build/lib.macosx-10.9-x86_64-cpython-311/mujoco/glfw
      copying mujoco/glfw/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/glfw
      running build_ext
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/private/var/folders/xf/3cnhbc2166j2zrtrl7_hnhtw0000gp/T/pip-install-j8hzc_ra/mujoco_23d0c118766242d3b273914e1da650c4/setup.py", line 292, in <module>
          setup(
        File "/Users/marktowers/opt/anaconda3/envs/Gymnasium3.11/lib/python3.11/site-packages/setuptools/__init__.py", line 87, in setup
          return distutils.core.setup(**attrs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Users/marktowers/opt/anaconda3/envs/Gymnasium3.11/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 185, in setup
          return run_commands(dist)
                 ^^^^^^^^^^^^^^^^^^
        File "/Users/marktowers/opt/anaconda3/envs/Gymnasium3.11/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
          dist.run_commands()
        File "/Users/marktowers/opt/anaconda3/envs/Gymnasium3.11/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 968, in run_commands
          self.run_command(cmd)
        File "/Users/marktowers/opt/anaconda3/envs/Gymnasium3.11/lib/python3.11/site-packages/setuptools/dist.py", line 1217, in run_command
          super().run_command(command)
        File "/Users/marktowers/opt/anaconda3/envs/Gymnasium3.11/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 987, in run_command
          cmd_obj.run()
        File "/Users/marktowers/opt/anaconda3/envs/Gymnasium3.11/lib/python3.11/site-packages/wheel/bdist_wheel.py", line 325, in run
          self.run_command("build")
        File "/Users/marktowers/opt/anaconda3/envs/Gymnasium3.11/lib/python3.11/site-packages/setuptools/_distutils/cmd.py", line 319, in run_command
          self.distribution.run_command(command)
        File "/Users/marktowers/opt/anaconda3/envs/Gymnasium3.11/lib/python3.11/site-packages/setuptools/dist.py", line 1217, in run_command
          super().run_command(command)
        File "/Users/marktowers/opt/anaconda3/envs/Gymnasium3.11/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 987, in run_command
          cmd_obj.run()
        File "/Users/marktowers/opt/anaconda3/envs/Gymnasium3.11/lib/python3.11/site-packages/setuptools/_distutils/command/build.py", line 132, in run
          self.run_command(cmd_name)
        File "/Users/marktowers/opt/anaconda3/envs/Gymnasium3.11/lib/python3.11/site-packages/setuptools/_distutils/cmd.py", line 319, in run_command
          self.distribution.run_command(command)
        File "/Users/marktowers/opt/anaconda3/envs/Gymnasium3.11/lib/python3.11/site-packages/setuptools/dist.py", line 1217, in run_command
          super().run_command(command)
        File "/Users/marktowers/opt/anaconda3/envs/Gymnasium3.11/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 987, in run_command
          cmd_obj.run()
        File "/private/var/folders/xf/3cnhbc2166j2zrtrl7_hnhtw0000gp/T/pip-install-j8hzc_ra/mujoco_23d0c118766242d3b273914e1da650c4/setup.py", line 154, in run
          self._mujoco_framework_path) = self._find_mujoco()
                                         ^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/xf/3cnhbc2166j2zrtrl7_hnhtw0000gp/T/pip-install-j8hzc_ra/mujoco_23d0c118766242d3b273914e1da650c4/setup.py", line 166, in _find_mujoco
          raise RuntimeError(
      RuntimeError: MUJOCO_PATH environment variable is not set
      [end of output]
```

Im uncertain of the exact error, my suspicion is that as a precompiled version doesn't exist, which `setup.py` assumes exists therefore fails.

saran-t, 2022-12-01:
We're planning to release 2.3.1 on Monday, which will be the first version that ships with prebuilt wheel for Python 3.11.

closed 2022-12-06 by kevinzakka

ForeverAT, 2023-09-30:
This problem still exists, 
File "/private/var/folders/3f/3rh85q_51cq806grwz6lqzgc0000gn/T/pip-install-k2japaq_/mujoco_3c68ed1334574fd3b245f88c545dee51/setup.py", line 156, in run
          self._mujoco_framework_path) = self._find_mujoco()
                                         ^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/3f/3rh85q_51cq806grwz6lqzgc0000gn/T/pip-install-k2japaq_/mujoco_3c68ed1334574fd3b245f88c545dee51/setup.py", line 170, in _find_mujoco
          raise RuntimeError(
      RuntimeError: MUJOCO_PATH environment variable is not set
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for mujoco
  Running setup.py clean for mujoco
Failed to build mujoco
ERROR: Could not build wheels for mujoco, which is required to install pyproject.toml-based projects

carloskiyoto, 2025-01-26:
Hi, I'm facing the same problem
Perhaps do you found how to fix it?
Thanks!