google-deepmind/mujoco #608: python introduction for beginner: cannot find gizmo.stl and attribute error

abbiesgame, 2022-11-27:
**How to ask for help**
Hi,

I'm a student and I'm trying to use MuJoCo for robotics reinforcement learning.

I'm looking for some help with python introductions.
I tried searching for answers in google and github issues but couldn't find them.
I know that I am missing something but I couldn't not have solved it myself after many attempts.
Even though, this is somewhat silly/easy question for many people in this community I wish my questions are answered.
Thank you for in advance for any reply.

Here is a model which explains my question:

<details>
I ran the code provided by mujoco official website python-bindings.
However, I keep getting errors.

1. FileNotFoundError: [Errno 2] No such file or directory: '/path/to/gizmo.stl'
2. AttributeError: partially initialized module 'mujoco' has no attribute 'MjModel' (most likely due to a circular import)

import mujoco

XML=r"""
<mujoco>
  <asset>
    <mesh file="gizmo.stl"/>
  </asset>
  <worldbody>
    <body>
      <freejoint/>
      <geom type="mesh" name="gizmo" mesh="gizmo"/>
    </body>
  </worldbody>
</mujoco>
"""

ASSETS=dict()
with open('/path/to/gizmo.stl', 'rb') as f:
  ASSETS['gizmo.stl'] = f.read()

model = mujoco.MjModel.from_xml_string(XML, ASSETS)
data = mujoco.MjData(model)
while data.time < 1:
  mujoco.mj_step(model, data)
  print(data.geom_xpos)

</details>

kbayes, 2022-11-29:
Hi,

1. You need to replace '/path/to/gizmo.stl' with an actually path name.
2. Sounds like an installation issue?

We just uploaded a new MuJoCo tutorial colab using the Python native bindings.  I recommend going through that as an introduction.  It's located in the Python subdir, but you can find the link in the README.

abbiesgame, 2022-11-29:
Thank you! @kbayes I will look in to it

abbiesgame, 2022-11-29:
I tried the tutorial on my computer but it still gives me error.
(Python verision 3.9.10; MuJoCo verision 2.3.0

`Traceback (most recent call last):
  File "C:\Users\ohyec\Documents\robotics\mujoco-2.3.0-windows-x86_64\mujoco.py", line 1, in <module>
    import mujoco
  File "C:\Users\ohyec\Documents\robotics\mujoco-2.3.0-windows-x86_64\mujoco.py", line 11, in <module>
    model = mujoco.mjModel.from_xml_string(xml)
AttributeError: partially initialized module 'mujoco' has no attribute 'mjModel' (most likely due to a circular import)`

The code I ran is:
`import mujoco

xml = """
<mujoco>
  <worldbody>
    <geom name="red_box" type="box" size=".2 .2 .2" rgba="1 0 0 1"/>
    <geom name="green_sphere" pos=".2 .2 .2" size=".1" rgba="0 1 0 1"/>
  </worldbody>
</mujoco>
"""
model = mujoco.mjModel.from_xml_string(xml)`

abbiesgame, 2022-11-29:
xml:
`<mujoco>
  <worldbody>
    <geom name="red_box" type="box" size=".2 .2 .2" rgba="1 0 0 1"/>
    <geom name="green_sphere" pos=".2 .2 .2" size=".1" rgba="0 1 0 1"/>
  </worldbody>
</mujoco>`

abbiesgame, 2022-11-29:
Also, I tried
`model = mujoco.mjModel.from_xml_string(xml)`
and
`model = mujoco.MjModel.from_xml_string(xml)`

abbiesgame, 2022-11-29:
I fixed it!
I named my python script as "mujoco" so switched to a unique name.