google-deepmind/mujoco #611: "Import mujoco" crashes on Macbook M1

kalfton, 2022-11-28:
Hi,

I am trying to install mujoco on my Macbook with M1 pro chip by running: 
pip install mujoco
It said "Successfully installed mujoco-2.2.0"

I also installed gym[mujoco]
And I found that "env = gym.make('Humanoid-v4')" crashed the python program with the error:
"zsh: illegal hardware instruction  python"

I investigate it further and found that when I try to import the mujoco package by running:
import mujoco
Python also crashed and gave me the same error

I tried to install mujoco-2.3.0, and I still have the same error.
Does anyone know why it happens and how to fix it?
Thank you

kevinzakka, 2022-11-29:
Hi, M1 user here: it works fine for me! Can you try installing MuJoCo on a fresh environment with python 3.7+ and seeing if the issue persists?

kalfton, 2022-11-29:
Hi Kevin,
I am using conda environment, and I tried to create new environments with python 3.7 and 3.9, but they still give me the same error.

I am a new Mac user though; one difference I notice is that when I run in python:
import platform
platform.platform()

It gives me the result:
'macOS-10.16-x86_64-i386-64bit'
rather than arm64, though the macbook uses M1.

Could this be a problem? But all the other packages including PyTorch work fine with it...

kevinzakka, 2022-11-29:
That's exactly the issue, you need to download Anaconda for arm64 and not x86!

kalfton, 2022-11-29:
Thank you Kevin; that is the problem. I reinstalled conda and it works!

closed 2022-11-29 by kalfton