google-deepmind/mujoco #613: How to rotate objects without 'freejoint' ??

miyukin73, 2022-11-30:
Hello, 

I want to rotate sphere objects **without 'freejoint'**.

I wrote about a sphere object by using 'hinge' as follows:

```
    <body name="particle_0" pos="-9.01593e-05 -0.014561 0.0702002" quat="0 0 0 0">
      <!-- <freejoint /> -->
      <joint axis="1 0 0" name="slide_x_p0" pos="0 0 0" type="slide" />
      <joint axis="0 1 0" name="slide_y_p0" pos="0 0 0" type="slide" />
      <joint axis="0 0 1" name="slide_z_p0" pos="0 0 0" type="slide" />
      <joint axis="1 0 0" name="rotate_x_p0" pos="0 0 0" type="hinge" />
      <joint axis="0 1 0" name="rotate_y_p0" pos="0 0 0" type="hinge" />
      <joint axis="0 0 1" name="rotate_z_p0" pos="0 0 0" type="hinge" />
      <!-- <joint axis="1 0 0" name="rotate_p0" pos="0 0 0" type="ball" /> -->
      <geom name="particle_0" type="sphere" size="0.0075" rgba="1 0 0 1" solref=".005 1" mass="0.005" contype="1" conaffinity="1" friction="0.5 0.5 0.5" />
    </body> 
```
However, the ball don't rotate as bellow.

https://user-images.githubusercontent.com/63595993/204781293-088edadd-1f09-4aa9-b0c8-deddce3a1132.mov

What is problem??

I attached some files regarding above movie.
[question.zip](https://github.com/deepmind/mujoco/files/10122594/question.zip)

Thank you.

Balint-H, 2022-11-30:
Interestingly, the problem arises from your rotation definition. All 3D rotations must be unit quaternions, and you define "0 0 0 0", which does not correspond to the identity rotation. Surprisingly, "0 0 0 0" is accepted as a rotation definition, but prevents all rotational transformations of the object in terms of the kinematic chain, but keeps track of it in joint space.  

If you remove that, and the unusually high friction values your particles can roll, however it is very unstable due to the low precision of the Euler angle representation (please correct me if that's wrong). If you really must avoid a free joint, you can get reasonable stability with a ball joint instead for rotations. If you need you can easily convert to Euler angles from ball joint's `qpos`, if thats what you need.

BTW why do you want to avoid a free joint in the first place?

miyukin73, 2022-11-30:
Hi, @Balint-H 

Thank you very much for the reply.

Then, you thought freejoint is the best option ?

The reason I wanted to try something other than **`freejoints`** is that I am currently having a problem with the sphere rotating **unnaturally** during cup rotation, and I thought that one of the causes might be related to **`freejoints`**.

----------------------------------------------------------------

I will explain the details in a little longer.
I am currently generating an environment with a cup that has a joint set only for rotation around the `x-axis` and a sphere that has a `freejoint` set.

And as you asked in #614, I am inputting the rotation speed of the cup every frame in the form `**data.qvel[0] = XX**`

Therefore, I'm not giving any input to the rotational movement of the sphere at all, but the sphere has rotated in place at a faster speed than the rotation of the cup, as shown in the following animation. (In this case, I only set `data.qvel[0] = 0` )

[Screencast from 2022年12月01日 03時36分22秒.webm](https://user-images.githubusercontent.com/63595993/204880803-87178f15-c154-4c75-98b0-5ffdaad65b73.webm)

In this case, the center of rotation of the cup (light blue arrow in the figure below) is set at the center of the bottom of the container.
[image]

----------------------------------------------------------------
[image]

Then, the center of rotation of the cup is set at the edge of the bottom surface as shown in the figure above, and when the center of rotation is moved closer to the sphere, it becomes as shown in the animation below.
[Screencast from 2022年12月01日 03時30分48秒.webm](https://user-images.githubusercontent.com/63595993/204879709-69ce9a41-b36e-4187-b78a-23a21ba59933.webm)

The sphere rotates unnaturally in the same way, but you can see from the movie that the rotation speed is slower than before.

----------------------------------------------------------------

Based on the above, we hypothesized that there might be a problem in the way the joints are set up, and were considering how to set up the joints without using free joints for the sphere.

I apologize very much for the general explanation, but do you have any suggestions for improvement regarding the above?
I am very sorry for repeating the question, but I would be grateful for your answer.

[question.zip](https://github.com/deepmind/mujoco/files/10126206/question.zip)

I am also attaching the program.
If you run **`main_1-1_rotate.py`**, you will be able to run the simulation.

Thank you in advance.

Balint-H, 2022-12-05:
Hello, this issue could come from your script somewhere (sorry didnt have time to test it), but if you assign a high damping to the cup you get a quasi-constant angular velocity. I've enabled the freejoint and added small marker geoms to visualise the particle's frame. If you just run the scene with simulate.exe, you can see that there is no unnatural rotation inside the cup.
[cup_edited.zip](https://github.com/deepmind/mujoco/files/10155956/cup_edited.zip) (Although I'd additionally recommend removing the quat assignment in the particles body definition in the xml)

Balint-H, 2022-12-06:
Heya, are you sure that 0 is the address of the container hinge velocity (and not one of the particle rotation velocities)?

If you are not sure you can always use named access to get it:

```python
# Get qpos from joint struct with named access
qpos_idx = self.data.joint('joint_name').qpos

# Alternatively get joint ID first, and get qpos index from that
joint_id = mujoco.mj_name2id(model, mujoco.mjtObj.mjOBJ_JOINT, 'joint_name')
qpos_idx = self.model.jnt_qposadr[joint_id]
```

Balint-H, 2022-12-06:
Heya, are you sure that 0 is the address of the container hinge velocity (and not one of the particle rotation velocities)?

If you are not sure you can always use named access to get it:

```python
# Get qpos from joint struct with named access
qpos_idx = model.joint('joint_name').dofadr  # Note that we need dofadr and not qposadr to get indexer for velocity

# Alternatively get joint ID first, and get qpos index from that
joint_id = mujoco.mj_name2id(model, mujoco.mjtObj.mjOBJ_JOINT, 'joint_name')
qvel_idx = model.jnt_dofadr[joint_id]
```

miyukin73, 2022-12-10:
Hi, @Balint-H 

Thank you very much for the new method regarding the speed command method.

I wrote a program as follows.
i)
```
qpos_idx = model.joint('rotate_x').dofadr
data.qvel[qpos_idx] = M
```

ii)
```
joint_id = mujoco.mj_name2id(model, mujoco.mjtObj.mjOBJ_JOINT, 'rotate_x')
qvel_idx = model.jnt_dofadr[joint_id]
data.qvel[qvel_idx] = M
```

But similarly, the granular object rotated unnaturally...

[Screencast from 2022年12月10日 14時25分46秒.webm](https://user-images.githubusercontent.com/63595993/206830907-63530f18-140b-4178-ac0c-2b8cb2f09908.webm)

I am very sorry to keep you busy, but I would greatly appreciate it if you could tell me if there is a way to make it work in python.

I attached the program.
[question.zip](https://github.com/deepmind/mujoco/files/10199268/question.zip)

Thank you.

Balint-H, 2022-12-10:
Heya, finally had the chance to have a look at the script in action. Here's a way to avoid this behaviour:

```python
for i in range(t_max):    
    dof_idx = model.joint('rotate_x').dofadr
    
    mujoco.mj_step1(model, data)
    data.qvel[dof_idx] = M
    mujoco.mj_step2(model, data)
```
It seems manual setting of `qvel` elements have an affect on bodies in contact with the joint with the edited velocity. If the body does not have a free joint, then it seems the manually edited `qvel` entries are overwritten inside [ `mj_step1`](https://mujoco.readthedocs.io/en/2.3.1/programming.html#simulation-loop).

You can edit e.g. hinge joint velocities directly between `mj_step1` and `mj_step2` as shown above. If you instead want to apply an impedance control law (based on your script it looks like), this would also be the right place to do it. For a more stable steady velocity rotation, you can apply damping D on the hinge, and set the torque T in `qfrc` between `mj_step1` and `mj_step2`. The steady state velocity will then be T/D (ignoring external forces). The larger T and D is, the less compliant is the movement to external perturbations.

(Also sorry for the mistaken `qpos_idx` variable name, should have changed it to something appropriate like `dof_idx`, fixed my original answer)

miyukin73, 2022-12-11:
Hello, @Balint-H 

Thank you very much for taking the time to check and answer our questions.

I now understand the cause of the unnatural rotation of the sphere when the container is rotated.
Thank you very much for your careful and easy-to-understand explanation.

Also, after applying the program you taught us, the unnatural rotation was eliminated and we were able to run an accurate simulation...!

Thank you so very much for taking the time in your busy schedule to respond to our request repeatedly !!

closed 2022-12-11 by miyukin73