google-deepmind/mujoco #614: [Basic Question] "data.qvel[0] = XX" is an appropriate way to specify objects' rotation velocity?

miyukin73, 2022-11-30:
Hello, 

I want to rotate a cup like bellow.

https://user-images.githubusercontent.com/63595993/204806691-ed76ec15-a06d-436a-a486-0374c10c4962.mp4

When I do the above, I wrote as follows:

```
model = mujoco.MjModel.from_xml_path(target_img_path)
data = mujoco.MjData(model)
data.qvel[0] = 10
```

Is this an appropriate way to specify rotational speed by `data.qvel[0] = XX` ?

Sorry for the basic question, but thank you in advance.

Balint-H, 2022-11-30:
That will set the velocity once, but it will change if you don't set it every frame. For something like this, also consider using a [mocap body](https://mujoco.readthedocs.io/en/stable/modeling.html?highlight=motion%20capture#mocap-bodies) to define its rotation every frame.

miyukin73, 2022-11-30:
I got it.

Thank you so much for your help!

closed 2022-11-30 by miyukin73