google-deepmind/mujoco #615: Is there any way to make adhesion active for contacts with specific object types?

tadashiK, 2022-11-30:
Hi,

I'm looking for some help with the adhesion actuator.

I was playing with the active adhesion example [here](https://github.com/deepmind/mujoco/tree/main/model/adhesion) and wondered if it is possible to make adhesion active for contacts with specific object types.
For example, the red box in this example is adhesive to everything. However, I am wondering if it is possible to make it adhesive only to green objects and not to gray crate.

Thank you in advance for any help!

tadashiK, 2022-11-30:
Just FYI, I want to make something like magnets. So, if there is any way to make magnets in MuJoCo, it would be nice to know it.

yuvaltassa, 2022-11-30:
If you disable contacts between two bodies using the regular methods (exclude or contype/affinity) then adhesion will not work, but they will also be able to penetrate.

If you want solid state magnets, which you will not be able to turn on/off using an actuator, that would be fairly easy to do. Is that what you want? Or is controllability important for your use case?

vaxenburg, 2022-11-30:
It seems like an interesting idea. Bodies could have a "stickiness" parameter, say between 0 and 1, that multiplies the adhesion force. The value of 1 would be equivalent to the current implementation.

tadashiK, 2022-11-30:
Hi Yuval,

Thanks a lot for your help!

> If you want solid state magnets, which you will not be able to turn on/off using an actuator, that would be fairly easy to do.

Yes, I need solid state magnets. An example of the thing I want to make is Magsafe cable of Mac. I am sorry for not detaling about what I want to make.

I am also sorry that I was aware of exclude but not contype. After reading the description about contype, I think I figuired out how to implement solid state magnets.

So, I think I need to set adapter's conaffinity to 2, and socket's contype/conaffinity to 2. Then, since adapter's contype is 1, the adapter still collides with usual objects, whose default conaffinity is 1, while the socket is adhesive only to the adapter (presicely speaking geom with contype/conaffinity=1).

Am I right, and this is how contype/affinity are supposed to be used?

tadashiK, 2022-11-30:
Hi Roman,

> It seems like an interesting idea. Bodies could have a "stickiness" parameter, say between 0 and 1, that multiplies the adhesion force. The value of 1 would be equivalent to the current implementation.

Oh, I didn't think about this idea, but I agree this is an interesting idea.

So, with this stickness parameter, for example, we can create weak and strong magnets. Am I right?

vaxenburg, 2022-11-30:
> So, with this stickness parameter, for example, we can adjust the magnetic susceptibility of objects (like metal has higher magnetic susceptibility while wooden objects have no magnetic susceptibility). Am I right?

Yup, this is exactly what I meant. It's a parameter of the body the magnet is acting on. It would also apply to modeling suction grippers and any other type of adhesion modeling.

yuvaltassa, 2022-11-30:
This is not how we would do it. Magnets can repel.

You would specify a magnet dipole axis and strength in physical units, I think best associated with a site, and then get the proper interaction wrench. 

Need to read up on magnetostatics but should be straightforward.

closed 2022-12-01 by tadashiK