google-deepmind/mujoco #621: Access the joint forces generated by `frictionloss`

studywolf, 2022-12-05:
Hello,

I'm a researcher trying to use MuJoCo for generating the parameters to control of a physical robot arm. We found that the Mujoco simulated arm movement best reflects the actual arm movement when we add `frictionloss` at each of the joints. I'm trying to figure out where in the code / if the effects of frictionloss on each joint are stored in the MjData are MjContact structs somewhere, so that I can use them to try to cancel out the friction effects during control.

I saw in the documentation https://mujoco.readthedocs.io/en/latest/computation.html?highlight=friction#friction-loss that it's used as a constraint, I'm hoping that doesn't mean it's not explicitly calculated and stored somewhere. I printed out a few of the friction related variables from the relevant structs (e.g. `data.contacts.friction`), but it wasn't clear to me how to process the output (e.g., `// tangent1, 2, spin, roll1, 2`). 
I'm hoping someone can provide insight on how I can retrieve the resulting friction forces applied to each joint so that I can negate them in my controller.
Thanks!

yuvaltassa, 2022-12-06:
`fricitonloss` is indeed a constraint, but unrelated to contacts. For example joint friction is not affected by the force through the joint, it is a simple dry friction force that opposes motion with a force possibly as large as, but no larger, than the specified frictionloss value.

You can find the actual force values by iterating through the `efc_force` vector. The ones you want have `efc_type == mjCNSTR_FRICTION_DOF`, and the associated joint id will be in `efc_id`.

closed 2022-12-06 by yuvaltassa

yuvaltassa, 2022-12-10:
`mujoco.mjtConstraint.mjCNSTR_CONTACT_FRICTIONLESS`