google-deepmind/mujoco #630: Streaming Motion capture data to a Mujoco model

tsgtdss583, 2022-12-07:
Hello There,
I am thinking of using Mujoco Python, however, i am super beginner. and I could not find many resources to solve my questions.
here is my issue.
I would like to input/stream a motion capture, which is in a txt/csv file to a Mujoco model. The documentation says to use the keyframe argument in MJCF and they will be coppied to mjData parameters when the simulation state is set to the keyframe. so my question is:
1, how can i set the keyframe MJCF argument to read data from a file? or do i need to copy the file data in to the XML?
2, how can i set the mjData parameteres to the keyframe data suring simulation? any example would help a lot.

kevinzakka, 2022-12-07:
Hi @tsgtdss583, I would start by reading about [mocap bodies](https://mujoco.readthedocs.io/en/latest/modeling.html?highlight=mocap#mocap-bodies) in the documentation.

yuvaltassa, 2022-12-09:
1. Yes. You can also save keyframes from within simulate and save the XML if that is more convenient.
2. `mj_resetDataKeyframe`