google-deepmind/mujoco #633: Do joints of type `free` support damping?

stefanscherzinger, 2022-12-07:
Hi,

I was wondering whether the `damping` parameter for joints of type `free` is supposed to have an effect?

## Some background
I built a model in which I want to steer a component in free space (I deactivated gravity) with forces and torques that come from a teach device. I'm applying the forces and torques via `mjData.xfrc_applied`. As is to expect, I get force-proportional acceleration. In my use case, operators should teleoperate the component (I'm ultimately interested in the feed forward strategies they use), which is why I would like to have some form of velocity-proportional damping.

However, the `damping` parameter does not seem to have an effect. Here's my model:

## My model
```xml
<mujoco model="simulator">
        <!--
        A simulator for teaching assembly by teleoperation.
        Users can steer the assembly components in free space and record their control commands and the simulator's states
        for further data processing. The recorded datasets can be used for Imitation Learning and (Offline) Reinforcement Learning.
        -->
        <compiler angle="radian"/>
        <size njmax="500" nconmax="100"/>
        <option>
                <flag gravity="disable" />
        </option>
        <!--
        <option viscosity="0.01"
                integrator="implicit" />
        -->
        <asset>

                <texture type="skybox" builtin="gradient" rgb1="0.9 0.9 0.9" rgb2="0.5 0.5 0.5" width="512" height="512"/>
                <texture name="texplane" type="2d" builtin="checker" rgb1=".25 .25 .25" rgb2=".3 .3 .3" width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>
                <material name="matplane" reflectance="0.2" texture="texplane" texrepeat="1 1" texuniform="true"/>

                <mesh name="passive_component" file="passive_component.stl"/>
                <mesh name="active_component" file="active_component.stl"/>
                <material name="passive_component" rgba="1.0 0.533 0.0 1.0"/>
                <material name="active_component" rgba="0.462 0.196 0.411 1.0"/>
        </asset>
        <worldbody>
                <light directional="true" diffuse=".8 .8 .8" specular=".2 .2 .2" pos="0 0 5" dir="0 0 -1" castshadow="false"/>
                <geom name="floor" pos="0 0 -0.10" size="0 0 1" type="plane" material="matplane"/>

                <!-- Fixed with the world -->
                <body name="passive_component" pos="0 0 0.0" quat="1 0 0 0">
                        <geom type="mesh" mesh="passive_component" material="passive_component"/>
                </body>

                <!-- Controllable with user commands. -->
                <body name="active_component">
                        <inertial pos="0 0 0" quat="1 0 0 0" mass="0.05" diaginertia="0.0001 0.0001 0.0001"/>
                        <joint name="free_joint" type="free" damping="1000"/>
                        <geom type="mesh" mesh="active_component" material="active_component"/>
                </body>
        </worldbody>

        <!-- Initial object poses for simulation -->
        <include file="initial_states.xml"/>

</mujoco>
```

I used joint damping with `hinge` joints in other models, which works great and would also be my preferred method for `free` joints.

I guess I could be using `mjModel.dof_damping[]` for this purpose, but I wanted to know if `type="free" damping="1000"` is supposed to do that and if so, how would it set the different Cartesian axes?

## Screenshot
[image]

kevinzakka, 2022-12-07:
Few comments:

1) njmax and nconmax are deprecated — check out the [memory](https://mujoco.readthedocs.io/en/latest/XMLreference.html#size)
2) Damping does apply to free joints unless you use a `freejoint` element. Test it out with just a free box in an empty scene.

stefanscherzinger, 2022-12-07:
@kevinzakka 

Thanks for your fast response! 

> Damping does apply to free joints unless you use a freejoint element.

Having this certainty led me to question my implementation again. I had a left-over in my code and overwrote `mjModel.dof_damping`'s entries. 

The damping element now works as expected. Thanks again.

closed 2022-12-07 by stefanscherzinger