google-deepmind/mujoco #639: simulate.exe GUI is rendered incorrectly all over the window

Juri-3D, 2022-12-09:
Hey,
I had to switch to a new computer and while the simulations are running as always, I can't use the simulate.exe anymore. 
The GUI is rendered all over the place (see screenshot). The buttons are still clickable on the left black panel, although not visible.

The PC:
ASRock X670E Pro RS
AMD Ryzen 5 7600X (with integrated graphics)
32GB RAM
M.2 SSD
AMD Radeon RX 6800 XT
Windows 11 22H2 (22621.819)

I've tried the latest AMD Adrenalin (22.5.1 and 22.11.1) and PRO (22Q4) graphics driver versions.
I've tried different mujoco releases (2.3.0, 2.3.1) and built it from the source code, always with the same result.
Any help is appreciated.

[image]

saran-t, 2022-12-09:
Can you try disabling DPI scaling?

Juri-3D, 2022-12-09:
> Can you try disabling DPI scaling?

I did, but nothing changed.
An addition to the specs above: I have two monitors, one with 1080p and one with 4k. But I already used those screens on the old PC.

saran-t, 2022-12-09:
Hard for me to make suggestions without being able to reproduce the issue. Are you comfortable with C programming? I'd recommend writing a minimal application that just compiles a model, sets up a rendering context, renders an image, dumps to file, and see if you can get something sensible.

Alternatively, if you can install Python, you can transcribe our [tutorial Colab](https://github.com/deepmind/mujoco/blob/main/python/tutorial.ipynb) into a small Python script (specifically, just only as far as the first image rendered) and see if you can get the correct image.

Juri-3D, 2022-12-20:
Sadly I'm a mechanical engineer with little experience in C or Python. 
The issue, however, seems not related to the simulation environment. I can open any example model or my complex models with the "basic.exe" and they are rendered and run perfectly fine. Only the clickable buttons/menus of the "simulate.exe" are broken.

Ocanath, 2023-02-21:
I have the same issue. New laptop with AMD Ryzen 5 PRO 6650U with Radeon Graphics, and up to date drivers (Adrenalin Edition Version 22.11.2).

I used the prebuilt simulate.exe binary of mujoco release 2.3.2. 
I also built simulate.exe from source, same exact issue as the release version.

[image]

I suspect this is an issue with AMD graphics specifically, since that seems to be the common element with me and @Juri-3D

Leo428, 2023-02-22:
This is happening for me as well. My laptop has AMD CPU and GPU (Lenovo Z16). I remember simulate.exe worked in late 2022.

saran-t, 2023-02-22:
@Leo428 Can I please ask you to please go through each [binary release version](https://github.com/deepmind/mujoco/releases) one-by-one to see when the error first occurred?

Leo428, 2023-02-22:
@saran-t 
I tried as far back as version 2.1.2 and the problem persists on my laptop (Windows 11 and AMD chips). I also tested it on my windows 10 PC with an intel CPU and Nvidia GPU and version 2.3.2 worked fine.

Since I reinstalled my laptop's system in late 2022, I guess it's most likely a driver issue caused by either windows 11 or AMD updates.

Ocanath, 2023-02-26:
Forgot to mention, I'm running windows 10

Shavdrag, 2023-04-01:
I had the same issue, but it worked after adding simulate.exe of the mujoco folder in the NVIDIA Control Panel program settings and set the preferred graphics processor to the NVIDIA GPU. 
I think the program tried to use the integrated graphics of my laptop and somehow that caused a problem

ArtifixPrime, 2023-04-04:
It seems to be a GPU driver issue. I also have a Radeon GPU and the issue resolved when I downgraded the driver. The version I'm currently using is Adrenalin 22.5.1

rajuptvs, 2023-04-10:
I am also having a similar issue on amd based cpu, but haven't faced the issue on my other intel based cpu.

saran-t, 2023-04-19:
I should've thought of this ages ago, but could people who are running into this problem please try setting the environment variables `MUJOCO_GL_DEBUG=1` and `LIBGL_DEBUG=verbose`, and see if you get any message?

saran-t, 2023-10-20:
I just got hold of a Windows AMD machine for the first time today, and while I haven't been able to reproduce anything nearly as garbled as the screenshots posted in this thread, I can nevertheless see that the UI panels are not aligned correctly.

Can I please ask anyone affected who's still following this thread to verify:
- that the Simulate app is still garbled, and
- what happens when you press Tab and Shift+Tab to hide both UI panes

Thanks!

NghiaPhamttk27, 2023-10-21:
The issue of the display being cut off still exists on my laptop. When I press Tab and Shift+Tab, the display returns to normal, but, of course, the user cannot interact with the UI panes. So, it is very likely that this issue is with the two interface panes. You can watch the video below.
[screen-capture (1).webm](https://github.com/google-deepmind/mujoco/assets/145578401/33576d3d-5263-45d5-a968-fdcaf55b5e1c)

saran-t, 2023-10-21:
Could you please patch the following, launch the viewer (both with and without a model loaded), and post screenshots here? This patch replaces the side panels with solid blocks of colours, hopefully this will help us better understand what operation is causing the problem.

[0001-Set-debug-colors.patch](https://github.com/google-deepmind/mujoco/files/13060734/0001-Set-debug-colors.patch)

NghiaPhamttk27, 2023-10-22:
I have tried both with a model and without a model. In the case with the model, even when I press Shift+Tab, I can't see the model as before. You can check the two videos below.
[With humanoid model.webm](https://github.com/google-deepmind/mujoco/assets/145578401/411a191a-0dba-49f7-953c-61833b5af03e)
[without model.webm](https://github.com/google-deepmind/mujoco/assets/145578401/67de0081-8e2a-436f-90c4-47dfeb3fa3cd)

saran-t, 2023-10-22:
Thanks. Unfortunately that's not at all what I expected to see (not that I know what I expected to see, but certainly not that!).

This means that I _still_ can't repro this problem fully. Might take a while to get this figured out.

NghiaPhamttk27, 2023-10-22:
Yeah, it's quite sad to hear. I still hope the issue can be resolved because I can't have an Nvidia GPU as a replacement anytime soon.

Xiaxuli, 2023-10-25:
[image]
help, why did I click simulate.exe and then the window flashed back

saran-t, 2023-10-28:
Interestingly, the new bug report #1144 shows an identical screenshot to what I saw on last week. Note that it's much less garbled than what's posted here thus far, but still wrong.

codeslord, 2023-11-14:
I have a laptop with AMD Internal GPU as well as Nvidia GPU running Windows 11. After recent AMD graphics driver update I faced the same issue. The way I fixed the issue is that, I went to Windows Settings->System->Display->Graphics and under Custom options for apps, clicked browse, chose the Simulate.exe. Once I have added the custom app, selected MuJoCo->Options and instead of let windows decide, chose my NVIDIA Graphics card as my Graphics Preference. Now it works fine.
[image]

Xiaxuli, 2023-11-14:
Thank you for your reply, but I tried your method, but it still did not solve the problem that the click could not open
[image]

rkourdis, 2023-12-23:
@saran-t I have the exact same issue as the screenshot in #1144 with AMD driver `23.12.1` (on Ryzen 7 6800U). This is on a local build using VS 2022.

Interestingly enough, downgrading to `22.5.1` fixes the GUI and everything renders well (I remember trying a couple of different driver versions a while ago). That might give some hints as to what's going on.

girishkolli, 2024-02-07:
I just started off with Mujoco and I faced the same problem. My CPU and GPU are both AMD. I believe is was on version 24.xx.xx.xx (forgot to check the exact number) and the GUI was glitching out. I downgraded my driver to 21.40.18.10 since it was already saved on my computer and now simulate.exe runs without any issues. I hope a fix can be found for the latest AMD drivers soon.

RajneeshSingh111, 2024-05-31:
> Hey, I had to switch to a new computer and while the simulations are running as always, I can't use the simulate.exe anymore. The GUI is rendered all over the place (see screenshot). The buttons are [...]

I faced the same issue with AMD Ryzen 6000 series and NVIDIA GEFORCE RTX GPU. I changed the GPU mode from standard to ultimate (in ASUS ROG zephyrus G15) and it fixed the issue as I shifted to NVIDIA GPU. I think there is any issue with the AMD driver.

wjxway, 2024-07-07:
Can confirm this issue is still there for AMD stand-alone graphics card (7900XTX). Hopefully it can be fixed soon.

moosewoler, 2024-11-11:
Same issue. Downgrade AMD driver to 21.12.1, then the problem was fixed.
My laptop works on AMD Ryzen 7 5700U with Radeon Graphics and Windows 11.

vivek-kumar9696, 2025-01-06:
> I have a laptop with AMD Internal GPU as well as Nvidia GPU running Windows 11. After recent AMD graphics driver update I faced the same issue. The way I fixed the issue is that, I went to Windows Settings->System->Display->Graphics and under Custom options for apps, clicked browse, chose the Simulate.exe. Once I have added the custom app, selected MuJoCo->Options and instead of let windows decide, chose my NVIDIA Graphics card as my Graphics Preference. Now it works fine. <img alt="image" width="523" src="https://private-user-images.githubusercontent.com/4618356/282788751-c387144f-9322-489e-9159-4e0bea5819fa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYxMzY2NzksIm5iZiI6MTczNjEzNjM3OSwicGF0aCI6Ii80NjE4MzU2LzI4Mjc4ODc1MS1jMzg3MTQ0Zi05MzIyLTQ4OWUtOTE1OS00ZTBiZWE1ODE5ZmEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDEwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMDZUMDQwNjE5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDA0MzBlZTYwZWZkYmZhYzkzMzNkYmFmNWM4NDI1ODBhYWNmOWUwYTFkNmE3NzYyNDA3OWE1ZDkwMzI2ZDY4YSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.lZ4qv_NJxZNuBKqF8A7_laSzjxEI6Hl1__eOfzNbzHQ">

This worked perfectly for me. Thanks!!

DIYman, 2025-03-13:
I can confirm that this issue still persists with Radeon RX 580, driver versions V24.9.1 and the previous release. The GUI is not functioning correctly—button clicks register inaccurately, as if there's an offset, causing unintended elements to be selected. Additionally, when a model is loaded, the GUI disappears entirely in simulate.exe.

AQLK, 2025-08-14:
Unfortunately, the/some problem still persists for menus for recent AMD drivers. It is not garbled like the top image in this thread, but it now mostly black and gray. I have added an mp4 to this comment (see bottom) that shows what it looks like. The 3D visualization and the scrollbars respond as expected to the mouse input. 
In the movie I am clicking _a lot_ on the black/gray parts where the menu should be. At some point something happens (something expanded?) so that I can then drag the scrollbar up/down to make the whole menu pass by. Nonetheless, none of the buttons work as they should, they get triggered when clicking somewhere else (exactly like DIYman above me says).
The menu is visible when the executable is started without a model, but buttons also responds incorrectly then, with vertical click offset.

(Also, scrolling and dragging the scrollbar behaves like tablet swiping (this inverted w.r.t. standard behavior on Windows, but might be common on Mac?).)

**Libraries (x64, Windows):**
Mujoco: 3.3.5, glfw: 3.4.

**Machine** (used AMD Adrenalin ensure latest driver for August 14):
[image]

https://github.com/user-attachments/assets/357af3a0-8315-4c8a-8f29-15a2183b92fa

DIYman, 2025-08-19:
Sadly, problem still persists for me too.

> As far as we can tell this has been fixed in recent versions of the AMD drivers. Can anyone confirm/deny that it is now solved?

I have tested with the latest available version of AMD driver (25.8.1) as of today and Mujoco 3.3.5. UI still doesnt work.

https://github.com/user-attachments/assets/52518537-d326-4709-86a4-5c6898095511

Diankes, 2026-02-22:
Same problem persist for me using a NVIDIA GeForce RTX 3050 Laptop GPU using windows 11, I tried changing windows Settings > Display > Graphics > Add a desktop app and adding the path to my binary either python.exe when using pip or the simulate.exe when using the release zip and changing the Gpu Preference to my GPU, same problem persisted after adding the binary path in NVIDIA Control Panel in Manage 3D Settings > Program Settings and adding the path to either python.exe or simulate.exe.

[image]

My env is
```plaintext
OS: Windows 11
GPU 1 (Integrated): AMD Radeon(TM) Graphics
  Driver: 31.0.12014.6 | Date: 7/11/2022 | WDDM 3.1
  VRAM: 495 MB | DirectX 12 Ultimate: Disabled

GPU 2 (Dedicated): NVIDIA GeForce RTX 3050 Laptop GPU
  Driver: 32.0.15.9186 | Date: 1/19/2026 | WDDM 3.2
  VRAM: 3962 MB | DirectX 12 Ultimate: Enabled
```

harishfaqot, 2026-04-27:
I have same problem here using AMD Ryzen 8840HS with Radeon 780 iGPU. The problem solved by updating the firmware of the driver using latest version. My current working version 26.2.2.

This is the link for my processor for example : https://www.amd.com/en/support/downloads/previous-drivers.html/processors/ryzen/ryzen-8000-series/amd-ryzen-7-8840hs.html

davidglickman2002-hue, 2026-10-24:
Ran into this issue as well with AMD Driver 31.0.21924.4004

The problem seems to be that the AMD OpenGL driver fails when using glBlitFramebuffer for the UI
I got it working locally with AI help by disabling UI multisampling and using glReadPixels and glDrawPixels instead.

If it helps for reference you can see my workaround: https://github.com/davidglickman2002-hue/mujoco/commit/09a39226cd8085032d6e45f2b952de5cce6fe7c7