google-deepmind/mujoco #643: Custom mujoco env with gym in RL (using the official pybinding - mujoco)

BaiYunpeng1949, 2022-12-11:
Hi,

I'm a PhD student from NUS-HCI lab, and I'm trying to use MuJoCo for customizing a gym environemnt that could be used in RL. I am using mujoco (not mujoco_py) + gym because I am extending the others' work. Please kindly find the work I am following [here](https://github.com/aikkala/user-in-the-box).

I'm looking for some help with How to start customizing simple environment inherited from gym, so that I can use their RL frameworks later. Should I just follow [gym's mujoco_env examples here](https://github.com/openai/gym/blob/6a04d49722724677610e36c1f92908e72f51da0c/gym/envs/mujoco/mujoco_env.py#L28)? 

To start with, I want to customize a simple env with an easy task, i.e., 2 planes and a moving dot. The dot could be considered as an agent, our target is letting it learn to move back and forth between the two planes instead of moving randomly, e.g., when it touches the first plane, it should be flying to the other plane as soon as possible.

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="2Planes">
  <option timestep="0.005"/>

  <visual>
    <map force="0.1" zfar="30"/>
    <rgba haze="0.15 0.25 0.35 1"/>
    <global offwidth="2560" offheight="1440" elevation="-20" azimuth="120"/>
  </visual>

  <statistic center="0 0 0.7"/>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1=".3 .5 .7" rgb2="0 0 0" width="32" height="512"/>
    <texture name="body" type="cube" builtin="flat" mark="cross" width="127" height="1278" rgb1="0.8 0.6 0.4" rgb2="0.8 0.6 0.4" markrgb="1 1 1" random="0.01"/>
    <material name="body" texture="body" texuniform="true" rgba="0.8 0.6 .4 1"/>
    <texture name="grid" type="2d" builtin="checker" width="512" height="512" rgb1=".1 .2 .3" rgb2=".2 .3 .4"/>
    <material name="grid" texture="grid" texrepeat="1 1" texuniform="true" reflectance=".2"/>
  </asset>

  <worldbody>

    <light name="top" diffuse="1 1 1" pos="0 0 5" dir="0 0 -1"/>

    <geom name="floor" size="0 0 .05" type="plane" material="grid" condim="3"/>
    
    <!-- Camera -->
    <!-- camera name="main" pos="0 0 0" euler="0 0 0" fovy="90"/ -->  # Right hand Cartesian coordinates. x--> right; y-->back; z-->up.

    <!-- Two rectangular planes -->
    <geom name="plane1" type="plane" pos="0.5 2 1.5" euler="75 0 0" size=".5 .5 0.1" rgba="1 0 0 1"/>   # Red.
    <geom name="plane2" type="plane" pos="-0.5 0 1.5" euler="75 0 0" size=".5 .5 0.1" rgba="0 1 0 1"/>  # Green.

    <!-- Moving dot or ball -->
    <body name="dot" pos="0 1 1">
      <freejoint/>
      <geom name="sphere" type="sphere" size="0.1" rgba="0 0 0.5 0.5"/>
    </body>
  </worldbody>
  
</mujoco>
```

</details>

Here is a screenshot / video, illustrating my question (red plane 1, green plane 2, and the blue freely moving dot): 
[image]

My codes could be found [here](https://colab.research.google.com/drive/1FqDM-G_9vVg4_VXdJj1LH8_QbzdoO7eX#scrollTo=xIuVsnBYIpFN).

I really appreciate your time and help!

saran-t, 2022-12-17:
Unfortunately we aren't able to help with the Gym API since that's not something that we develop or use. You'd be better off asking on https://github.com/Farama-Foundation/Gymnasium.

Having said that, the Gymnasium team seems to have done a good job at porting environments to the new `mujoco` Python bindings. If you want to follow examples, their `v4` environments (e.g. [this one](https://github.com/Farama-Foundation/Gymnasium/blob/main/gymnasium/envs/mujoco/humanoid_v4.py)) might be a good starting point.

closed 2022-12-17 by saran-t

denisgriaznov, 2024-04-19:
I encountered the same problem, and not finding a direct tutorial, I made a simple example of creating a Gymnasium environment from my own MuJoCo model.
 
https://github.com/denisgriaznov/CustomMuJoCoEnviromentForRL 

If you have any questions, write!

BaiYunpeng1949, 2024-04-19:
To whom it may concern:

You may refer to https://github.com/Synteraction-Lab/heads-up-multitasker to see how to use MuJoCo + gym + RL for simulation. And I am happy to help if you need it.

denisgriaznov, 2024-04-19:
> > I encountered the same problem, and not finding a direct tutorial, I made a simple example of creating a Gymnasium environment from my own MuJoCo model.   [https://github.com/denisgriaznov/CustomMuJoCoEnviromentForRL ](https://github.com/denisgriaznov/CustomMuJoCoEnviromentForRL%C2%A0)
> > If you have any questions, write!
> 
> Hi thanks for the sharing but your link is invalid for me, is it a private repo?
> 
> Thank you for reminding, I finally solved the issue with my senior mentor, and here is the repo: https://github.com/Synteraction-Lab/heads-up-multitasker We published a paper on that on CHI '24, Heads-Up Multitasker: Simulating Attention Switching On Optical Head-Mounted Displays

Sorry, there were extra spaces at the end of the link :)
Edited the link, it is now available.
Glad it worked out for you, great job!