google-deepmind/mujoco #644: `import torch` causes segfault when setting `MUJOCO_GL=osmesa`

vwxyzjn, 2022-12-12:
**How to submit a good bug report**

```python
# g.py
from dm_control import suite
import numpy as np
import torch # this causes segmentation fault
# Load one task:
env = suite.load(domain_name="cartpole", task_name="balance")

# Iterate over a task set:
for domain_name, task_name in suite.BENCHMARKING:
  env = suite.load(domain_name, task_name)

# Step through an episode and print out reward, discount and observation.
action_spec = env.action_spec()
time_step = env.reset()
while not time_step.last():
  action = np.random.uniform(action_spec.minimum,
                             action_spec.maximum,
                             size=action_spec.shape)
  time_step = env.step(action)
```
```
MUJOCO_GL=osmesa CUDA_VISIBLE_DEVICES="-1" python g.py
free(): invalid pointer
[1]    3991540 IOT instruction (core dumped)  MUJOCO_GL=osmesa CUDA_VISIBLE_DEVICES="-1" python g.py
```

Here is a screencast:

[image]

- Include the following context:
  - Operating system: 
```
             /////////////                costa@pop-os 
         /////////////////////            ------------ 
      ///////*767////////////////         OS: Pop!_OS 21.10 x86_64 
    //////7676767676*//////////////       Kernel: 5.17.5-76051705-generic 
   /////76767//7676767//////////////      Uptime: 29 days, 2 hours, 59 mins 
  /////767676///*76767///////////////     Packages: 3038 (dpkg), 6 (flatpak), 2 (snap) 
 ///////767676///76767.///7676*///////    Shell: zsh 5.8 
/////////767676//76767///767676////////   Terminal: node 
//////////76767676767////76767/////////   CPU: AMD Ryzen 9 3900X (24) @ 3.800GHz 
///////////76767676//////7676//////////   GPU: NVIDIA GeForce RTX 3060 Ti 
////////////,7676,///////767///////////   GPU: NVIDIA GeForce RTX 3060 Ti 
/////////////*7676///////76////////////   Memory: 5493MiB / 64237MiB 
///////////////7676////////////////////
 ///////////////7676///767////////////                            
  //////////////////////'////////////                             
   //////.7676767676767676767,//////
    /////767676767676767676767/////
      ///////////////////////////
         /////////////////////
             /////////////

```
  - MuJoCo version (and if the bug is new, the version where it used to work). `'2.3.1'`
  - For Python issues, what bindings are you using (i.e `mujoco`, `dm_control`, `mujoco-py`)? `dm_control`

saran-t, 2022-12-16:
I'm unable to reproduce this issue on Debian. Can you please run this in a debugger (e.g. gdb) and post a stack trace at the segfaulting point?

saran-t, 2022-12-16:
OK, looks like the error only arises during `atexit`, i.e. when Python is exiting. I'm getting a SIGABRT rather than a SIGSEGV on my machine though.

```
Thread 1 "python" received signal SIGABRT, Aborted.
__pthread_kill_implementation (threadid=<optimized out>, signo=6, no_tid=no_tid@entry=0) at ./nptl/pthread_kill.c:44
44	./nptl/pthread_kill.c: No such file or directory.
(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=6, no_tid=no_tid@entry=0) at ./nptl/pthread_kill.c:44
#1  0x00007fa1128895df in __pthread_kill_internal (signo=<optimized out>, threadid=<optimized out>) at ./nptl/pthread_kill.c:89
#2  __GI___pthread_kill (threadid=<optimized out>, signo=<optimized out>) at ./nptl/pthread_kill.c:89
#3  0x00007fa1129f5e70 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#4  0x00007fa112828469 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007fa11287d888 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7fa1129b66fb "%s\n") at ../sysdeps/posix/libc_fatal.c:155
#6  0x00007fa11289322a in malloc_printerr (str=str@entry=0x7fa1129b41d0 "free(): invalid pointer") at ./malloc/malloc.c:5659
#7  0x00007fa112894d6c in _int_free (av=<optimized out>, p=<optimized out>, have_lock=have_lock@entry=0) at ./malloc/malloc.c:4434
#8  0x00007fa1128978df in __GI___libc_free (mem=<optimized out>) at ./malloc/malloc.c:3386
#9  0x00007fa06b8cbaa9 in ?? () from /lib/x86_64-linux-gnu/libLLVM-14.so.1
#10 0x00007fa11283ffb5 in __run_exit_handlers (status=0, listp=0x7fa1129f4818 <__exit_funcs>, run_list_atexit=run_list_atexit@entry=true, 
    run_dtors=run_dtors@entry=true) at ./stdlib/exit.c:113
#11 0x00007fa11284012a in __GI_exit (status=<optimized out>) at ./stdlib/exit.c:143
#12 0x00007fa112829211 in __libc_start_call_main (main=main@entry=0x55bc3819bb80, argc=argc@entry=2, argv=argv@entry=0x7ffc8fbeeee8)
    at ../sysdeps/nptl/libc_start_call_main.h:74
#13 0x00007fa1128292bc in __libc_start_main_impl (main=0x55bc3819bb80, argc=2, argv=0x7ffc8fbeeee8, init=<optimized out>, fini=<optimized out>, 
    rtld_fini=<optimized out>, stack_end=0x7ffc8fbeeed8) at ../csu/libc-start.c:389
#14 0x000055bc3819bab1 in _start ()
```

saran-t, 2022-12-16:
Importing `torch` before `mujoco` seems to remove the crash. Suspecting bad interaction between libLLVM and CUDA.

closed 2023-04-30 by kevinzakka