google-deepmind/mujoco #645: Size mismatch of mjd_transitionFD

Beanpow, 2022-12-12:
The nq is not equal to nv of the free joint. The nq of free joint is 7 and the nv is 6, because of the unit quaternion.
You can use the following minimal case to test.
```
<mujoco model="test">
  <worldbody>
    <body name="base" pos="0 0 0">
      <geom type="capsule" size="0.045" fromto="0 0 0  0.5 0 0"/>
      <joint type="free" />
    </body>
  </worldbody>
</mujoco>
```

**However**, some mujoco functions suppose they are equal, such as mjd_transitionFD.
So I can't use the function correctly.

erez-tom, 2022-12-12:
The code **doesn't** assume `nv==nq` blindly.  Instead, remember that we're computing **derivatives** of positions, and the space of delta-position really has dimensionality `nv` (even when positions have dimensionality of `nq!=nv`).  These Jacobians compute quantities whose semantics are delta-position.  If you need to add a vector of delta-positions computed using these matrices to a vector of positions, you could use [mj_integratePos](https://github.com/deepmind/mujoco/blob/main/include/mujoco/mujoco.h#L442).

closed 2022-12-12 by erez-tom

yuvaltassa, 2022-12-12:
You might want to look at our [LQR tutorial](https://colab.research.google.com/github/deepmind/mujoco/blob/main/python/LQR.ipynb) for a worked example.

Beanpow, 2022-12-13:
Thanks! @erez-tom @yuvaltassa