google-deepmind/mujoco #655: Simulation step counter in `mjData`

vaxenburg, 2022-12-15:
A very simple feature request: please consider adding a simulation step counter to the `mjData` data structure.

The simulation time `mjData->time` is there, but sometimes it is also useful to have an integer step counter. Of course, the current step is just time/timestep, but it's cleaner to just keep track of it directly in `mjData`.

saran-t, 2022-12-15:
Can you elaborate on what you need this for?

vaxenburg, 2022-12-15:
For example, in training the agent to track mocap data, like [here](https://arxiv.org/abs/1911.06636) for instance, the current simulation step is needed to get the current reference pose, etc.

The current solution is, of course, to maintain a counter as in [this](https://github.com/deepmind/dm_control/blob/main/dm_control/locomotion/tasks/reference_pose/tracking.py#L220) tracking task, but I think it would be cleaner to just have it in `physics.data`.

yuvaltassa, 2022-12-15:
On the one hand, this is very easy to implement, no strong objection.

On the other hand, here are two objections 🙂 
1. If the timestep is fixed, `round(data.time/model.timestep)` will just give you this number.
2. If the timestep is not fixed, for example if someone implements a variable-timestep integrator or changes the timestep mid-simulation, this number could be confusing/misleading.

What are your objections to using `round(data.time/model.timestep)`?

saran-t, 2022-12-15:
>  if someone ... changes the timestep mid-simulation

Concretely, we initialisers that steps physics in order to settle things then reset the time back to zero. I'd actually object fairly strongly to this proposal: we shouldnt introduce a second variable that effective tracks time but that may (and I'd argue, likely to) lose sync with the actual time when interacting with user code.

vaxenburg, 2022-12-15:
I agree that the possibility of loosing sync is bad enough to drop this idea.

closed 2022-12-15 by vaxenburg