google-deepmind/mujoco #659: Maximum number of objects

lukass-code, 2022-12-16:
Hello,

my question is about the maximum number of objects that Mujoco can handle.
Currently I have the problem that the simulation works fine with up to 300 objects.
But when I try to load a model with 400 or more objects, mujoco crashes.
Is there a maximum number of objects that mujoco can handle?

Thanks for your answer.

saran-t, 2022-12-17:
Can you please post the model that you're trying to simulate and the error that you're getting?

lukass-code, 2022-12-18:
Thanks for your answer,
thats the error messange:

```
Sun Dec 18 19:11:32 2022
ERROR: stack overflow: max = 299869096, available = 97857896, requested = 106218000 (ne = 0, nf = 0, nefc = 3988, ncon = 668)

```
Here is the working model: [Does work_txt.txt](https://github.com/deepmind/mujoco/files/10254876/Does.work_txt.txt)
This model causes the error: [Does not work_txt.txt](https://github.com/deepmind/mujoco/files/10254875/Does.not.work_txt.txt)
(I had to upload .txt files, because GitHub doesn't support .xml)

lukass-code, 2022-12-19:
After another search I came across the following tags:
[xml reference-size](https://mujoco.readthedocs.io/en/stable/XMLreference.html#size)

Using
```
 <size memory="2G" />
```
fixed it

closed 2022-12-19 by lukass-code