google-deepmind/mujoco #662: What should I put in `MUJOCO_PLUGIN_PATH` when building the python binding from source?

breakds, 2022-12-17:
Hi,

I'm an engineer and I'm trying to use MuJoCo for a research project on quadruped robot.

I'm looking for some help with building the latest MuJoCo python binding from source. I am following [this part](https://github.com/deepmind/mujoco/tree/main/python#source) of the README.md. It seems that this section is outdated as in the last step, `setup.py` requires not only `MUJOCO_PATH` but also `MUJOCO_PLUGIN_PATH` to be set. See [here](https://github.com/deepmind/mujoco/blob/main/python/setup.py#L168).

I was not able to find documentation explaining what this environment variable means. Can you help give some pointers?

Thanks a lot!

Waybaba, 2023-01-04:
> 

have you found any solutions?

breakds, 2023-01-04:
> > 
> 
> have you found any solutions?

No I haven't. I gave up building from source and use the wheel instead. But it would still be great if someone can shed some light on this.

saran-t, 2023-01-04:
You can check out [our CI config](https://github.com/deepmind/mujoco/blob/main/.github/workflows/build.yml) to see how we've set it up.

breakds, 2023-01-04:
> You can check out [our CI config](https://github.com/deepmind/mujoco/blob/main/.github/workflows/build.yml) to see how we've set it up.

Thanks! So basically it points to the plugin output of the compilation of the c++ library. Somehow my grep setup ignores the dot directories and the search for the keyword did not show `.github` content up.

breakds, 2023-01-04:
> Thanks! So basically it points to the plugin output of the compilation of the c++ library. Somehow my grep setup ignores the dot directories and the search for the keyword did not show `.github` [...]

Or prebuilt plugin binaries put in that directory I think.

closed 2023-01-04 by breakds