google-deepmind/mujoco #663: How to pin the cloth on the capsule

Kakcalu13, 2022-12-19:
Hi,

I'm learning how to use Mujoco and I'm trying to use MuJoCo for fun/study.

I'm looking for some help with joint part. I tried to follow the documentation but it seems too overwhelmed for me to start somewhere. I'm more of visual learner. (Not a great way to describe, I admit) 

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="2D grid with pins">
  <include file="scene.xml"/>

  <size nconmax="500" njmax="1000" nstack="10000000"/>

  <worldbody>
    <body name="test" mocap="true" pos="0 1 0.2">
      <geom type="capsule" size=".1 .1" material="matmarble" group="1" condim="1"/>
        <body>
            <joint type="hinge" pos="0 1 0.2"/>
            <composite type="grid" count="10 11 1" spacing="0.05" offset="0 0 1">
              <skin rgba=".6 .1 .6 1" inflate="0.001" subgrid="4"/>
              <pin coord="2 0"/>
              <pin coord="8 0"/>
              <geom size=".01"/>
            </composite>
        </body>
    </body>
  </worldbody>
</mujoco>
```

</details>

Here is a screenshot / video, illustrating my question:

I am trying to put a cloth on the capsule so I can see if cloth moves like a cape as I move the capsule.

See the screenshot here:
[image]

Left is the reality of result. The right is the goal. I wanted to put the cloth (a cape) on the capsule. 

using joint on composite seems a little too confusing (BIG for me).

Kakcalu13, 2022-12-21:
I've updated my code:

```
<mujoco model="2D grid with pins">
  <include file="scene.xml"/>
  <size nconmax="1000" njmax="10000" nstack="10000000"/>

  <worldbody>
    <gravity value="0 0 -9.81" />
    <body name="test" pos="0 1 0.2">
      <geom type="capsule" size=".1 .1" material="matmarble" group="1" condim="1"/>
       <joint type="slide" axis="1 0 0"/>
      <composite type="grid" count="5 11 1" spacing="0.05" offset="0 0.17 0.1">
        <skin rgba=".6 .1 .6 1" inflate="0.001" subgrid="4"/>
        <pin coord="0 0"/>
        <pin coord="4 0"/>
        <geom size=".01"/>
      </composite>
    </body>
  </worldbody>
</mujoco>

```

It's almost on the capsule which is good enough for me.

I have another question, how do you make it not "enter" capsule like this screenshot?
[image]

quagla, 2023-01-03:
You could try `solref="0.01 1" solimp=".95 .99 .0001"` in the `geom` inside the `composite` and see if that helps with avoiding penetration. Likely you will still get some. Let me know if that helps.

closed 2023-01-03 by quagla

Kakcalu13, 2023-01-03:
Thanks! I will test it tonight!