google-deepmind/mujoco #664: Calculate global orientation of ant legs

cowolff, 2022-12-19:
Hi,

I'm a student and I'm trying to use MuJoCo for Reinforcement Learning.

I'm looking for some help with calculating the global rotation of objects. I am using the ant model, but I need to read out the global rotation of the legs of the ant in python, but somehow I can't really get it.

For example, how would I get the global rotation of the geom left_ankle_geom? I tried the various data fields I can access from data.geom(i) and model.geom(i) but I can't figure it out. I need the global rotation, so I can export each frame of the latest training epoch to a JSON file, which I can then parse into a different 3D Engine for better visualization.

```XML
<worldbody>
    <body name="torso" pos="0 0 3">
      <camera name="track" mode="trackcom" pos="0 -3 0.3" xyaxes="1 0 0 0 0 1"/>
      <site name="box_range" type="box" pos="0.4 0 0" size=".05 .05 .05" rgba="0 .9 0 1" euler="0 90 0"/>
      <geom name="torso_geom" pos="0 0 0" size="0.25" type="sphere"/>
      <joint armature="0" damping="0" limited="false" margin="0.01" name="root" pos="0 0 0" type="free"/>
      <body name="front_left_leg" pos="0 0 0">
        <geom fromto="0.0 0.0 0.0 0.2 0.2 0.0" name="aux_1_geom" size="0.08" type="capsule"/>
        <body name="aux_1" pos="0.2 0.2 0">
          <joint axis="0 0 1" name="hip_1" pos="0.0 0.0 0.0" range="-30 30" type="hinge"/>
          <geom fromto="0.0 0.0 0.0 0.2 0.2 0.0" name="left_leg_geom" size="0.08" type="capsule"/>
          <body pos="0.2 0.2 0">
            <joint axis="-1 1 0" name="ankle_1" pos="0.0 0.0 0.0" range="30 70" type="hinge"/>
            <geom fromto="0.0 0.0 0.0 0.4 0.4 0.0" name="left_ankle_geom" size="0.08" type="capsule"/>
          </body>
        </body>
      </body>
</worldbody>
```

The export funtion
```Python
def export_json(model, data, filename):
    """
    Export the model and data to a json file
    """
    export_dict = {}
    geoms = [data.geom(i).name for i in range(model.ngeom)]
    bodies = [data.body(i).name for i in range(model.nbody)]
    for i in range(len(geoms)):
        xquat = data.geom(i).xmat
        # xquat = model.geom(i).quat
        export_dict[i] = {
            "pos": list(data.body(i).xpos),
            "rotation": list(mat2eulerScipy(xquat)),
            "size": list(model.geom(i).size),
            "type": int(model.geom(i).type[0]),
            "name": str(model.body(i).name),
        }
    json_object = json.dumps(export_dict, indent=4)
    with open(filename, "w") as outfile:
        outfile.write(json_object)
```

aravindsiv, 2022-12-19:
I believe `scipy` uses the quaternion ordering `x,y,z,w`, while mujoco uses `w,x,y,z`. That could be a reason the returned orientations look incorrect.

cowolff, 2022-12-20:
Thanks for both of your replies. I was unaware of the scipy problem, but because I used matrix rotation that wasn't the problem (I know the variable was still named xquat, sorry for that!). Now I took both of your comments and used the data.body(i).xquat and scipy with the correct order to calculate the angles, but the problem still persists and I think it has something to do with the different joints.

Quick example:
This is a small piece of my environment where I connected 2 boxes with a hinge:
```XML
<body pos="-0.934 -3.053 .488" name="TestAngle" euler="0 0 45">
      <geom conaffinity="1" condim="3" type="box" size="1.5 0.05 0.5" rgba="0 .9 0 1"/>
      <body pos="-0.75 -3 .488" name="Test2">
        <joint type="hinge" axis="0 0 1" range="-90 90" damping="0.1"/>
        <geom conaffinity="1" condim="3" type="box" size="0.05 1.5 0.5" rgba="0 .9 0 1"/>
      </body>
</body>
```
```Python
def quat2eulerScipy(quat):
    quat = [quat[1], quat[2], quat[3], quat[0]]
    r =  Rotation.from_quat(quat)
    angles = r.as_euler("zyx",degrees=True)
    return angles
```

Now TestAngle and Test2 have very different global rotations (see Screenshot), but when I export the rotations with my quat2eulerScipy function, both come out as having 45.0, 0.0, 0.0. This is correct for the TestAngle, but not for Test2.

[image]

I then tried the same using data.geom(i).xmat with my mat2eulerScipy() function, but I again got the same results. Both came out having 45.0, 0.0, 0.0 which doesn't make sense. So no matter whether I access the geom itself or the body, I don't get the correct global orientation. It is probably something fairly obvious, but I am still pretty new to MuJoCo.

```Python
def mat2eulerScipy(mat):
    mat = mat.reshape((3,3))
    r =  Rotation.from_matrix(mat)
    angles = r.as_euler("zyx",degrees=True)
    return angles
```

cowolff, 2022-12-20:
Okay, I am at this point going to post my entire code in here. I have this ant environment and somehow the global orientations are calculated correctly for the TestBox and for the Test2Box, but not for the legs of the ant. I think it has something to do with the angles of the joints, but I am not sure how to handle this right now.

And if I don't comment out the line in the python code, the angles for the ant are calculated correctly, but the angles for test boxes are wrong.

```Python
def mat2eulerScipy(mat):
    mat = mat.reshape((3,3))
    r =  Rotation.from_matrix(mat)
    euler = r.as_euler("zyx",degrees=True)
    # euler = np.array([-1 * euler[0], -1 * euler[1], -1 * euler[2]])
    return euler
```

```XML
<mujoco model="ant">
  <compiler angle="degree" coordinate="local" inertiafromgeom="true"/>
  <option integrator="RK4" timestep="0.01"/>
  <custom>
    <numeric data="0.0 0.0 0.55 1.0 0.0 0.0 0.0 0.0 1.0 0.0 -1.0 0.0 -1.0 0.0 1.0" name="init_qpos"/>
  </custom>
  <default>
    <joint armature="1" damping="1" limited="true"/>
    <geom conaffinity="0" condim="3" density="5.0" friction="1 0.5 0.5" margin="0.01" rgba="0.8 0.6 0.4 1"/>
  </default>
  <asset>
    <texture builtin="gradient" height="100" rgb1="1 1 1" rgb2="0 0 0" type="skybox" width="100"/>
    <texture builtin="flat" height="1278" mark="cross" markrgb="1 1 1" name="texgeom" random="0.01" rgb1="0.8 0.6 0.4" rgb2="0.8 0.6 0.4" type="cube" width="127"/>
    <texture builtin="checker" height="100" name="texplane" rgb1="0 0 0" rgb2="0.8 0.8 0.8" type="2d" width="100"/>
    <material name="MatPlane" reflectance="0.5" shininess="1" specular="1" texrepeat="60 60" texture="texplane"/>
    <material name="geom" texture="texgeom" texuniform="true"/>
  </asset>
  <worldbody>
    <light diffuse=".5 .5 .5" pos="0 0 3" dir="0 0 -1" />
    <geom conaffinity="1" condim="3" type="plane" size="40 40 1.110223E-16" rgba=".9 0 0 1" />
    <body pos="-0.934 -3.053 .488" name="TestAngle" euler="0 0 45">
      <geom conaffinity="1" condim="3" type="box" size="1.5 0.05 0.5" rgba="0 .9 0 1" name="TestBox"/>
      <body pos="-0.934 -3.053 .512" name="Test2">
        <geom conaffinity="1" condim="3" type="box" size="1.5 0.05 0.5" rgba="0 .9 0 1" name="Test2Box"/>
      </body>
    </body>
    <body name="torso" pos="0 0 3">
      <camera name="track" mode="trackcom" pos="0 -3 0.3" xyaxes="1 0 0 0 0 1"/>
      <site name="box_range" type="box" pos="0.4 0 0" size=".05 .05 .05" rgba="0 .9 0 1" euler="0 90 0"/>
      <geom name="torso_geom" pos="0 0 0" size="0.25" type="sphere"/>
      <joint armature="0" damping="0" limited="false" margin="0.01" name="root" pos="0 0 0" type="free"/>
      <body name="front_left_leg" pos="0 0 0">
        <geom fromto="0.0 0.0 0.0 0.2 0.2 0.0" name="aux_1_geom" size="0.08" type="capsule"/>
        <body name="aux_1" pos="0.2 0.2 0">
          <joint axis="0 0 1" name="hip_1" pos="0.0 0.0 0.0" range="-30 30" type="hinge"/>
          <geom fromto="0.0 0.0 0.0 0.2 0.2 0.0" name="left_leg_geom" size="0.08" type="capsule"/>
          <body pos="0.2 0.2 0">
            <joint axis="-1 1 0" name="ankle_1" pos="0.0 0.0 0.0" range="30 70" type="hinge"/>
            <geom fromto="0.0 0.0 0.0 0.4 0.4 0.0" name="left_ankle_geom" size="0.08" type="capsule"/>
          </body>
        </body>
      </body>
      <body name="front_right_leg" pos="0 0 0">
        <geom fromto="0.0 0.0 0.0 -0.2 0.2 0.0" name="aux_2_geom" size="0.08" type="capsule"/>
        <body name="aux_2" pos="-0.2 0.2 0">
          <joint axis="0 0 1" name="hip_2" pos="0.0 0.0 0.0" range="-30 30" type="hinge"/>
          <geom fromto="0.0 0.0 0.0 -0.2 0.2 0.0" name="right_leg_geom" size="0.08" type="capsule"/>
          <body pos="-0.2 0.2 0">
            <joint axis="1 1 0" name="ankle_2" pos="0.0 0.0 0.0" range="-70 -30" type="hinge"/>
            <geom fromto="0.0 0.0 0.0 -0.4 0.4 0.0" name="right_ankle_geom" size="0.08" type="capsule"/>
          </body>
        </body>
      </body>
      <body name="back_leg" pos="0 0 0">
        <geom fromto="0.0 0.0 0.0 -0.2 -0.2 0.0" name="aux_3_geom" size="0.08" type="capsule"/>
        <body name="aux_3" pos="-0.2 -0.2 0">
          <joint axis="0 0 1" name="hip_3" pos="0.0 0.0 0.0" range="-30 30" type="hinge"/>
          <geom fromto="0.0 0.0 0.0 -0.2 -0.2 0.0" name="back_leg_geom" size="0.08" type="capsule"/>
          <body pos="-0.2 -0.2 0">
            <joint axis="-1 1 0" name="ankle_3" pos="0.0 0.0 0.0" range="-70 -30" type="hinge"/>
            <geom fromto="0.0 0.0 0.0 -0.4 -0.4 0.0" name="third_ankle_geom" size="0.08" type="capsule"/>
          </body>
        </body>
      </body>
      <body name="right_back_leg" pos="0 0 0">
        <geom fromto="0.0 0.0 0.0 0.2 -0.2 0.0" name="aux_4_geom" size="0.08" type="capsule"/>
        <body name="aux_4" pos="0.2 -0.2 0">
          <joint axis="0 0 1" name="hip_4" pos="0.0 0.0 0.0" range="-30 30" type="hinge"/>
          <geom fromto="0.0 0.0 0.0 0.2 -0.2 0.0" name="rightback_leg_geom" size="0.08" type="capsule"/>
          <body pos="0.2 -0.2 0">
            <joint axis="1 1 0" name="ankle_4" pos="0.0 0.0 0.0" range="30 70" type="hinge"/>
            <geom fromto="0.0 0.0 0.0 0.4 -0.4 0.0" name="fourth_ankle_geom" size="0.08" type="capsule"/>
          </body>
        </body>
      </body>
    </body>
  </worldbody>
</mujoco>
```

saran-t, 2022-12-20:
When you said that the orientation is wrong, I assume that you have some bit of code that produces the expected orientation. Please give us both the code that produces the wrong and the expected orientations.

cowolff, 2022-12-20:
Okay, I am at this point going to post my entire code in here. I have this ant environment and somehow the global orientations are calculated correctly for the TestBox and for the Test2Box, but not for the legs of the ant. I think it has something to do with the angles of the joints, but I am not sure how to handle this right now.

And if I don't comment out the line in the python code in the mat2eulerScipy function, the angles for the ant are calculated correctly, but the angles for test boxes are wrong.

Python code:
```Python
def mat2eulerScipy(mat):
    mat = mat.reshape((3,3))
    r =  Rotation.from_matrix(mat)
    euler = r.as_euler("zyx",degrees=True)
    # euler = np.array([-1 * euler[0], -1 * euler[1], -1 * euler[2]])
    return euler

def export_json(model, data, filename, xml_path):
    """
    Export the model and data to a json file
    """
    export_dict = {}
    for i in range(model.ngeom):
        xmat = data.geom(i).xmat
        xpos = data.body(i).xpos
        export_dict[i] = {
            "pos": list(xpos),
            "quat": list(mat2eulerScipy(xmat.reshape((3,3)))),
            "size": list(model.geom(i).size),
            "type": int(model.geom(i).type[0]),
            "name": str(model.geom(i).name),
        }
    json_object = json.dumps(export_dict, indent=4)
    with open(filename, "w") as outfile:
        outfile.write(json_object)

def start_sim(export = True, frame = 0, max_frame = 200):
    xml_path = 'envs/ant.xml'
    model = mj.MjModel.from_xml_path(xml_path)  # MuJoCo model
    data = mj.MjData(model)                     # MuJoCo data

    while frame < max_value:
        time_prev = data.time
        
        while (data.time - time_prev < 1.0/60.0):
            mj.mj_step(model, data)
        frame = frame + 1
        if export:
            export_json(model, data, "exports/last" + str(frame) + ".json", xml_path)

start_sim()
exit()
```

Mujoco file:
```XML
<mujoco model="ant">
  <compiler angle="degree" coordinate="local" inertiafromgeom="true"/>
  <option integrator="RK4" timestep="0.01"/>
  <custom>
    <numeric data="0.0 0.0 0.55 1.0 0.0 0.0 0.0 0.0 1.0 0.0 -1.0 0.0 -1.0 0.0 1.0" name="init_qpos"/>
  </custom>
  <default>
    <joint armature="1" damping="1" limited="true"/>
    <geom conaffinity="0" condim="3" density="5.0" friction="1 0.5 0.5" margin="0.01" rgba="0.8 0.6 0.4 1"/>
  </default>
  <asset>
    <texture builtin="gradient" height="100" rgb1="1 1 1" rgb2="0 0 0" type="skybox" width="100"/>
    <texture builtin="flat" height="1278" mark="cross" markrgb="1 1 1" name="texgeom" random="0.01" rgb1="0.8 0.6 0.4" rgb2="0.8 0.6 0.4" type="cube" width="127"/>
    <texture builtin="checker" height="100" name="texplane" rgb1="0 0 0" rgb2="0.8 0.8 0.8" type="2d" width="100"/>
    <material name="MatPlane" reflectance="0.5" shininess="1" specular="1" texrepeat="60 60" texture="texplane"/>
    <material name="geom" texture="texgeom" texuniform="true"/>
  </asset>
  <worldbody>
[... 350 lines cut ...]
    "15": {
        "pos": [
            0.3999999999999997,
            -0.39999999999999947,
            0.3724809872182389
        ],
        "quat": [
            -18.436586232116202,
            -37.76230079223358,
            -50.77115336258751
        ],
        "size": [
            0.08,
            0.28284271247461906,
            0.0
        ],
        "type": 3,
        "name": "fourth_ankle_geom"
    }
}
```

So the angles for the legs are wrong and when I then import that JSON file into my Unreal Engine, it pretty much looks like that: So I think you can see that the legs are pretty weird.

[image]

nimrod-gileadi, 2024-01-02:
Apologies if this is still an issue for you. Closing since there hasn't been any activity on this issue in a while.

closed 2024-01-02 by nimrod-gileadi