google-deepmind/mujoco #665: Using the built-in python viewer as a non-interactive synchronized viewer?

breakds, 2022-12-21:
Hi,

I really like the new `viewer` in the python bindings that wraps `simulate.cc`. If my understanding is correct, it is currently serving as an interactive viewer for the users to interact with the models.

I am wondering whether I can use it for another use case. I would like to visualize how a neural network plays with the robot model. In this case, the viewer is in a "view only" mode, and it updates the visualization as the `MjModel` and `MjData` step forward with the actions produced by the neural network. As a plus, if I can still use the mouse to change the viewpoint in the viewer it would be great.

Currently when I call `viewer.launch(model, data)` it will block. If it is started in another process, I am not sure what is the recommended approach to send the update on the `MjData` to that process.

Thanks!

yuvaltassa, 2022-12-21:
I believe you are describing the passive viewer mode described in the 3rd subsection [here](https://mujoco.readthedocs.io/en/latest/python.html#interactive-viewer).

breakds, 2022-12-21:
Thanks for the prompt answer! Actually I tried the 3rd subsection approach, which is `viewer.launch_repl`. It will start the viewer, but still blocking the original process, and it also shows a python REPL prompts. Should I do this in a different process?

breakds, 2022-12-21:
If I understand the code logic [here](https://github.com/deepmind/mujoco/blob/main/python/mujoco/viewer.py#L264) for `launch_repl` correctly, it still blocks the main process by waiting for the repl thread to join. The is is undesired in the use case I am looking at because the main process need to be used for the actual interaction between Mujoco and the neural network.

yuvaltassa, 2022-12-21:
I see, you want a non-blocking passive viewer, correct?

I'm not sure you can do that in a single process in Python, but I might be wrong. @saran-t, is there a recommended MO for this use case?

breakds, 2022-12-21:
> I see, you want a non-blocking passive viewer, correct?

Yes exactly. Sorry I did not make it clear enough.

breakds, 2022-12-21:
I have also tried to put `viewer.launch` in a separate thread:

```
def _run_viewer(model, data):
    viewer.launch(model, data, run_physics_thread=False)

self._viewer_thread = threading.Thread(target=_run_viewer, args=(self._mj_model, self._mj_data))
self._viewer_thread.start()
```

So that `simulate.renderloop()` within `launch()` is caged in the thread. The viewer can start, but it does not render anything.

cemoke, 2022-12-21:
> 

In m1 mac using jupyter notebook, even the viewer does not start for me inside the thread

saran-t, 2022-12-21:
Running the viewer in a thread is problematic, that's why the REPL mode is implemented the way it is.

For the use case that you mentioned, the best thing to do would be to run your agent loop in a separate thread, but then you'd need to do the timekeeping yourself (you almost definitely don't want to just let the loop run as fast as possible), or install your controller into `mjcb_control` and let the viewer handle the timing as usual.

aftersomemath, 2022-12-21:
[Here](https://github.com/aftersomemath/mujoco/blob/e1ba2e6e2220758539f034b5a5ddbcdfdb4c6e34/model/double_pendulum/double_pendulum.py) is an example of using the `mjcb_control` callback to add a controller to the physics simulation run by the viewer (at first glance this sounds sufficient for what you want to do). You can still inject perturbations with the mouse, pause, reload, etc, and the camera can still be moved. [Here](https://www.youtube.com/watch?v=ZEnhaMDs6pw) is a video.

[Here](https://github.com/deepmind/mujoco/blob/6a58fd1522e0a78644e4cae86bc8ec377e0a4730/python/mujoco/viewer.py#L71) is the viewer's physics thread which you could re-implement so that your code controls every aspect of the physics simulation (and thus you can also add a controller).

closed 2022-12-21 by breakds