google-deepmind/mujoco #666: How to set mujoco library flag for makefiles in v2.3.1?

dtorre38, 2022-12-21:
Hello,

I have updated mujoco from 2.2.1 to 2.3.1, but now all my C makefiles are broken. The issue stems from the library flag for the  mujoco framework (shown in the screenshot below). Did the library call change for 2.3.X versions of mujoco? For 2.2.1 it was:

`LIBS = -framework mujoco`

but that doesn't work anymore. The issue is resolved if I downgrade the mujoco app back to v2.2.1, but that doesn't get me the latest software. Any suggestions on how to fix the library call for 2.3.1 would be greatly appreciated.

Here is an example makefile that no longer runs on 2.3.1: 

`GLFWROOT?=/usr/local/Cellar/glfw/3.3.8
MUJOCOPATH?=/Applications/MuJoCo.app/Contents/Frameworks

CFLAGS = -O2 -F$(MUJOCOPATH) -I$(GLFWROOT)/include -pthread
ALLFLAGS = $(CFLAGS) -L$(GLFWROOT)/lib -Wl, -rpath, $(MUJOCOPATH)
LIBS = -framework mujoco -lglfw
CC = gcc

ROOT = control_pendulum

all:
	$(CC) $(ALLFLAGS) main.c $(LIBS) -o $(ROOT)

main.o:
	$(CC) $(COMMON) -c main.c

clean:
	rm *.o `

Here is a screenshot of the error:

[image]

kevinzakka, 2022-12-22:
FYI, the Makefiles aren't being maintained -- the recommended way to build is via CMake. Is there any reason you are not using that btw?

yuvaltassa, 2022-12-22:
Concretely, we are leaning towards removing the makefiles altogether.

So if you have a clear reason to prefer raw makefiles over cmake please let us know so that we can take it into account.

dtorre38, 2023-01-02:
No particular reason for not using CMake. I found it easy to create a makefile for different simulations.

closed 2023-01-11 by dtorre38