google-deepmind/mujoco #670: Help installing MuJoCo on FPGA

alvinli6, 2022-12-23:
Hi,

I'm a student researcher and I'm looking for some help installing MuJoCo on an FPGA.

Here is a screenshot of the error I'm getting when I run pip install mujoco:

[image]

Does anyone have any idea how to fix this? Thank you so much in advance! I'm pretty new to interacting with the GitHub community, I apologize if I posted this in the wrong place.

saran-t, 2022-12-23:
Can you please give us more detail on exactly what platform you're trying to build for?

alvinli6, 2022-12-23:
@saran-t Hello, thanks for getting back to me! I’m trying to install on the PYNQ-Z2 board.

saran-t, 2022-12-27:
I know very little about FPGAs, however I'd be very surprised if you can get MuJoCo to run on the board...

If nothing else, the MuJoCo Python package is just bindings into C code, there's very little actual Python code. Having said that, if you want to try it, please consult [our documentation](https://mujoco.readthedocs.io/en/latest/python.html#building-from-source) for instructions on how to build from source.

yuvaltassa, 2022-12-27:
Also, while the bindings have some fairly modern C++, the MuJoCo engine itself is pure ANSI C and will compile on anything, including FPGA. If you decide you can do without Python, we can help you compile MuJoCo itself.

kevinzakka, 2023-01-16:
Closing for now, @alvinli6 feel free to re-open if you run into newer issues.

closed 2023-01-16 by kevinzakka