google-deepmind/mujoco #671: Static Magnet for Connectors

tadashiK, 2022-12-23:
First of all, thank you very much for all contributors to this MuJoCo project. This is an awesome repository.

**Is your feature request related to a problem? Please describe.**

I want to make a magnet to implement something like [MagSafe 3](https://www.apple.com/shop/product/MPL23AM/A/usb-c-to-magsafe-3-cable-2-m-space-gray) of Mac.

Implementing magnetic forces might be useful for other functions too. For example, adhesion currently attracts all objects within margin - gap with the same force strength. However, if one needs a realistic suction gripper, its force strength needs to depend on the distance between the gripper and an object, in a way similar to magnetic forces. (Closer, stronger.)

**Describe the solution you'd like**

It would be nice if we can add to an object a sclar attribute (like "magnetism"), which determines the magnitude of the force, and another attribute that determines the orientation of the magnetism (S and N). It would be also nice if setting the orientation to some special value allows to instantiate something like metal-like objects (no repulsion, no attraction of other objects, only attraction to magnets) and gravitation (no repulsion, only attraction). That said, for the purpose I descibed above (i.e., MagSafe 3), ferromagnetism and magnetic forces caused by electric currents are unnesesary.

**Describe alternatives you've considered**

Currently, I have no other idea on how to implement magnets.  Setting contype and conaffinity with adhesion actuators allows an ad-hoc method to implement something like magnets. However, it sometimes leads to no contacts between some objects.

**Additional context**

I opened an issue about this topic, and Yuval asked me to open a feature request. See this issue for the additional context: https://github.com/deepmind/mujoco/issues/615#issuecomment-1332673517

brilliant-ember, 2023-09-17:
Yeah this will be a great addition. Passive magnets or even electromagnets can add a lot of potential.
@yuvaltassa  do you think it's easy enough to implement?

yuvaltassa, 2023-09-17:
Yes, this should be straightforward to implement. Unfortunately we are very busy with other things and it's hard to imagine that this will be top priority. If you are up for it we can discuss what would be the best API (I have an idea) and you could implement and send a PR?

brilliant-ember, 2023-09-17:
Yes of course, I'd love to contribute!

yuvaltassa, 2023-09-17:
Okay, let's talk features and then once we're happy with them, we'll talk API.

In terms of features, here's what I think we **might* want, some of this should probably be excluded. Comments welcome. @saran-t, @quagla, @kbayes, your input in particular.
1. Unsigned monopoles with force $\propto r^{-2}$ to simulate Newtonian gravity. 
2. Signed monopoles with force $\propto r^{-2}$ in the direction of the sign product to simulate Coulomb electrostatic force.
3. Signed dipoles to simulate fixed magnets. Note that dipole interaction is roughly $\propto r^{-4}$ (I think? My memory is blurry) and involves torques as well as forces.
4. Actuated signed dipoles to simulate electromagnets.
5. Unsigned monopoles that interact with the dipoles in two ways: interaction of charged particles with the magnetic field a-la Maxwell, interaction of ferromagnetic materials with magnetic fields. One could also imagine dielectric materials I suppose? This entire item might be out of scope.
6. Ways of specifying the interaction constants (gravitational constant, permittivity, susceptibility, permeability etc.) such that reasoning about units is possible.

In terms of API all of these will likely be specified as point source at sites, but let's leave that until we agree on the feature set.

@brilliant-ember do you have the required physics background to follow the above?

brilliant-ember, 2023-09-18:
Last time I saw a physics textbook was few years ago. I would need a *big* refresher

saran-t, 2023-09-18:
We might as well allow a vector of multipole moments (equiv. spherical harmonics coefficients) to be specified for each body. That's more principled.

ryznir, 2024-09-05:
Has there been any update on this?

yuvaltassa, 2024-09-05:
Not really. It's not a very requested feature...

I'll make a deal with you. If you find me the equations for dipole-dipole interaction, I will make the feature 🙂

ryznir, 2024-09-05:
I'm more interested in having objects that emit a magnetic flux, for use with a magnetometer (to represent a Hall-Effect sensor)

yuvaltassa, 2025-03-18:
You're right, I hadn't thought about sensors, but clearly if you're defining fields in space you should be able to sense them.

<digression>
Interestingly though, gravity is unsensable. That's sort of a three-word summary of General Relativity. E.g. an accelerometer will output zeros along any ballistic trajectory. We think of ballistic trajectories as curved lines; actually they are straight lines in a spacetime which is curved.
</digression>