google-deepmind/mujoco #672: How to get link-side/output force for an actuator?

rohanpsingh, 2022-12-24:
Hi,

I'm a student and I'm trying to use MuJoCo for simulating humanoid robots.

I'm sorry if this API already exists for this and I just missed it.
I want to calculate the link-side or output force applied by any actuator (aka the "joint torque").
I guess one way to do this would be to sum up all the frictional torques (coming from `damping` and  `frictionloss`) on the joint and add it to the `qfrc_applied`. Is there any other, more direct way to accomplish this?

Thanks a lot to all the maintainers an contributors of this awesome project!

yuvaltassa, 2022-12-24:
Sounds like you want a [force sensor](https://mujoco.readthedocs.io/en/latest/XMLreference.html#sensor-force)?

rohanpsingh, 2022-12-24:
Thanks! Indeed, now that I think about it, a force/torque sensor placed appropriately at the joint should do the job. Let me try to find if some publicly available models have already implemented this.

closed 2022-12-24 by rohanpsingh