google-deepmind/mujoco #673: Help modeling vehicular collisions

aravindsiv, 2022-12-24:
Hi,

I'm a student trying to use MuJoCo to model vehicular dynamics in an environment with static obstacles.

I'm looking for some help with my model, particularly with detecting collisions.

Here is a model which explains my question:

```XML
<mujoco>
	<option gravity="0 0 -9.81" integrator="RK4" timestep="0.005" />
	<worldbody>
        <light diffuse=".5 .5 .5" pos="0 0 3" dir="0 0 -1"/>
		<geom name="floor" type="plane" size="5 5 0.1" rgba=".9 .9 .9 1"/>
        <geom name="obstacle" type="box" size=".25 .25 .25" pos="1.0 0 0.1" rgba="0 0 1 1"/>
		<body name = "chassis" pos="0 0 0.2" euler='0 90 0'>
			<joint type="free"/>
			<geom type="box" size=".05 .2 .5" rgba=".9 .9 0 1"/>
			<site name="marker" pos = "0 0 0.1" size="0.1" />
			<body name="left-tire" pos="0 0.3 -0.5" euler='90 0 0'>
				<joint name = "left-wheel" type="hinge" axis="0 0 -1"/>
				<geom type="cylinder" size=".2 0.05" rgba="0 .9 0 1"/>
			</body>
			<body name="right-tire" pos="0 -0.3 -0.5" euler='90 0 0'>
				<joint name = "right-wheel" type="hinge" axis="0 0 -1"/>
				<geom type="cylinder" size=".2 0.05" rgba="0 .9 0 1"/>
			</body>
		</body>
	</worldbody>
	<sensor>
		<framepos objtype="site" objname="marker"/>
	</sensor>
	<actuator>
		<velocity name="left-velocity-servo" joint="left-wheel" kv="100"/>
		<velocity name="right-velocity-servo" joint="right-wheel" kv="100"/>
	</actuator>
</mujoco>
```

When I run the following code:

```C++
int ncon = d->ncon;
if (ncon > 0)
{
    for (int i = 0; i < ncon; i++)
    {
        auto s1 = mj_id2name(m, mjOBJ_BODY, m -> geom_bodyid[d->contact[i].geom1]);
        auto s2 = mj_id2name(m, mjOBJ_BODY, m -> geom_bodyid[d->contact[i].geom2]);
        std::cout << "Contact " << i << ": " << s1 << " " << s2 << std::endl;
    }

}
```

it prints out:
```
Contact 0: world chassis
Contact 1: world chassis
Contact 2: world left-tire
Contact 3: world left-tire
Contact 4: world right-tire
Contact 5: world right-tire
```

Why are there multiple collisions between the same pair of bodies? Or since I'm defining the geoms for the floor and the obstacle inside the `worldbody` tag, collisions with them are registered as collisions with the worldbody?

What is the best practice to model acceptable pairs of collisions (such as between the floor and the vehicular body)? The documentation on how to do this is not very clear on this.

Thank you!

saran-t, 2022-12-27:
A cylinder-plane geom pair can generate up to 2 contact points, specifically when the cylinder lies sideways on the plane a contact point is generated on each end of the intersecting line segment. Similarly, box-box can generate up to 8 points.

In your case I'd recommend modeling the tires as capsules rather than cylinders.

aravindsiv, 2023-01-25:
Thanks, modeling wheels as ellipsoids seems to work for now!

closed 2023-01-25 by aravindsiv