google-deepmind/mujoco #674: Is there any way to create a composite model from import obj/stl file?

KunZHANG1994, 2022-12-24:
I would to import a existing obj file as soft model.
I have tried the composite model, but not work. 
I would like to know is there any way to create a composite model from import .obj/.stl file?
Does current version of Mujoco supports import a existing obj file as soft model?

quagla, 2023-01-03:
Hi, that's not supported yet. Is your aim to have a 3d soft body (i.e. tetrahedral mesh) or a surface (like a cloth)?

KunZHANG1994, 2023-01-04:
> Hi, that's not supported yet. Is your aim to have a 3d soft body (i.e. tetrahedral mesh) or a surface (like a cloth)?

Hi, thanks for your reply. I would like to simulate a 3D space closed surface, not a single piece of cloth. I have made a mesh file of cloth from blender. Is that possible? or Mujoco will further support?

quagla, 2023-01-09:
Yes it will be possible in the next release but with the caveat that the equilibrium configuration for the bending stiffness will be assumed to be the flat configuration (an angle pi across triangles). For a closed surface probably it won't matter that much because you could try setting the bending stiffness to zero since the deformations are dominated by stretching.

KunZHANG1994, 2023-01-09:
> 
Cool, very exciting news. By the way, when will be the next release available?

quagla, 2023-01-11:
Probably in a couple of weeks, but if you build from source you might be able to access the new feature earlier.

KunZHANG1994, 2023-01-11:
OK！ Thanks！

quagla, 2023-02-07:
We just made a new release but unfortunately the feature you requested did not make the deadline. You should keep an eye for it in the next month. I hope this does not change your plans in using MuJoCo.

KunZHANG1994, 2023-02-07:
> @KunZHANG1994 We just made a new release but unfortunately the feature you requested did not make the deadline. You should keep an eye for it in the next month. I hope this does not change your plans in using MuJoCo.

OK, Thx for telling me these. I changed my working schedule and waiting your new feature update.

Shamp0o, 2023-03-29:
Is this supported, yet? Couldn't find anything related to this in the elease notes of 2.3.3 and neither could I find a pull request.

quagla, 2023-04-17:
We are working on a larger feature that will include this, so I'm afraid it has been delayed. Apologies for this.

dblanm, 2023-08-14:
@quagla  any update on when will this be released?

quagla, 2023-08-14:
Unfortunately it won't be part of the next release. I've had this code ready for a while but unfortunately it's pending on a `composite` refactoring that has been delayed.

quagla, 2023-08-30:
Hi @dblanm and @KunZHANG1994 happy to let you know that this is now supported! Let me know if it works as you expected.

closed 2023-08-30 by quagla