google-deepmind/mujoco #675: defining velocity and/or power limits for actuators

lok-i, 2022-12-25:
Hi,

I'm a PhD student, and I'm trying to use MuJoCo to simulate our custom-designed bipedal robot.

I'm looking for some help with defining velocity/ power limits for actuators, especially of type motor. While the motor tag already has attributes to limit the ctrl, length and force, I couldn't find anything to limit the velocity/power. As you may know, electric motors are characterized by their torque and angular velocity curve, where torque x angular velocity = power (constant) and often have an explicit higher angular velocity limit aswell. What is the best way for me to insert these actuator constraints into my simulation, preferably through the model XML file? 

For example, the motor we use is from [unitree](https://shop.unitree.com/products/unitree-a1-motor), where peak torque = 33 Nm and peak angular velocity = 21 rad/s. Thus I would like the above values to be always below the said limits in simulation, yet allow my control policies to come up with strategies where it can trade-off between high angular velocity and high torque to respect the power limit (i.e. to be <= theoretical peak power, VI - losses)  in accordance to the task requirements.

Thanks!

rohanpsingh, 2022-12-25:
I have the same question too! Thus far, I've been defining the actuator velocity limits in the `user` attribute and then manually handling it in my code. But yes, this is probably not ideal.

yuvaltassa, 2022-12-25:
My memory of this is a bit fuzzy, so please correct me if I'm wrong, but is this not a result of back-EMF?

As the motor velocity increases in one direction, a current is formed which resists the motion. The maximum velocity you describe is *not an independent constraint*, but simply *that velocity at which the generated back-EMF current equals the actuation current* (for a simple linear motor). If this is the case, then there is a simple solution: add **damping** to your joint or actuator.

I.e. setting  `<joint damping="d"/>` where `d = peak_torque / peak_velocity` will give you exactly the correct effect. Note that I'm using joint damping assuming that you are using the Euler integrator (which only integrates joint damping implicitly). If you are using the implicit integrator, you can equivalently use `<actuator biastype="affine" biasprm="0 0 -d"/>`.

Please follow up here if any of this doesn't make sense (or is wrong 🙂)

closed 2022-12-25 by yuvaltassa

Krishnendu1984, 2023-06-07:
Sir I am working in Mujoco earlier I tried walking a biped with reinforcement learning with actuator values -0.3 to 0.3 and gear 20.,with 11 kg mass .But when I scale up my model 4 times how can I modify these actuator values to make it walk so is there any formulae to find ctrl values and gear values

yuvaltassa, 2023-06-07:
You need to use dimensional analysis for this. Try it first with a simplified model

Krishnendu1984, 2023-06-07:
So sir is it like that gear ratio multiplied by ctrllimit give torques from - values to + values?

Uptomylimit, 2023-08-28:
Hi! So if I use the rk4 integrator, can I set the damping as you said to get this result of back-EMF?

yuvaltassa, 2023-08-28:
Yes. I would not recommend RK4 though, `implicitfast` is better at integrating damping terms. RK4 is better for energy-conserving systems.