google-deepmind/mujoco #677: Gripping of Cable by Fin Gripper

makitayugo, 2022-12-28:
Hi,

I'm a student and I'm trying to use MuJoCo for Master's research.

I'm looking for some help with my model, Grasping Cables in Flexible Object Manipulation Tasks.

Here is a model which explains my question:

```XML
<mujoco model="panda scene">

  <include file="panda.xml"/>

  <statistic center="0.3 0 0.4" extent="1"/>
  <extension>
    <required plugin="mujoco.elasticity.cable"/>
  </extension>

  <visual>
    <headlight diffuse="0.6 0.6 0.6"  ambient="0.3 0.3 0.3" specular="0 0 0"/>
    <rgba haze="0.15 0.25 0.35 1"/>
    <global azimuth="120" elevation="-30"/>
  </visual>
  <size nconmax="500" njmax="1000"/>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512"
        height="3072"/>
    <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4"
        rgb2="0.1 0.2 0.3" markrgb="0.8 0.8 0.8" width="300" height="300"/>
    <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5"
        reflectance="0.2"/>
  </asset>

  <worldbody>
    <light pos="0 0 1.5" dir="0 0 -1" directional="true"/>
    <geom name="floor" size="0 0 0.05" type="plane" material="groundplane"/>
    <composite type="cable" curve="s" count="51 1 1" size="2" offset="-1 0.3 .2"  initial="none">
      <plugin plugin="mujoco.elasticity.cable">
        <!--Units are in Pa (SI)-->
        <config key="twist" value="1e6"/>
        <config key="bend" value="1e5"/>
      </plugin>
      <joint kind="main" damping=".5"/>
      <geom type="capsule" size=".01" rgba=".8 .2 .1 1" friction="4" condim="1"/>
    </composite>
    <!--<body name="socket" pos="1 0.3 .1">
      <geom type="box" size="0.05 0.05 0.05" margin=".02" gap=".01" rgba=".1 .8 .2 1" contype="2" conaffinity="2"/>
    </body>-->
    <body name="box1" pos="0.1 0.3 .1" zaxis="0 0 1">
      <geom type="box" size="0.05 0.05 0.05"  group="1" condim="1"/>
    </body>
    <body name="box2" pos="0.5 0.3 .1" zaxis="0 0 1">
      <geom type="box" size="0.05 0.05 0.05"  group="1" condim="1"/>
    </body> 
  </worldbody>
</mujoco>```

This xml file is intended for use with the robot arm "panda" to grasp a cable. However, when it grabs and lifts the cable, the cable slips off.
```<geom type="capsule" size=".005" rgba=".8 .2 .1 1" friction="2" condim="1"/>```
I have included the parameter :friction to indicate friction, but I am not sure if it is working.

Here is a video of the implementation.

[Screencast from 2022年12月19日 16時53分27秒.webm](https://user-images.githubusercontent.com/85473041/209772926-cef63c9c-3730-487a-b506-e81897977a99.webm)

quagla, 2023-01-03:
Hi @makitayugo, `condim="1"` means no friction, see https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-geom, try using 3 and let me know if that works.

closed 2023-01-09 by quagla

makitayugo, 2023-01-12:
I can't grab it after changing to ``condim="3"``
We believe that the arm is not gripping enough.
If there is a way to do this, we would like to know.
This xml file is based on ``scene.xml``, shown below.
https://github.com/deepmind/mujoco_menagerie/blob/main/franka_emika_panda/scene.xml