google-deepmind/mujoco #678: Obtaining cable location information

makitayugo, 2022-12-28:
Hi,

I'm a student and I'm trying to use MuJoCo for Master's research.
I'm looking for some help with Obtaining cable location information.

I am trying to write code in Python to get the location of a cable, but I don't know how to do it. I would like to know if there is a function or other function that can be used for this.

quagla, 2023-01-03:
Hi, the cable is noting but a collection of bodies (try to export the XML after you load a cable, you will see that the composite is expanded into multiple bodies). So accessing their positions is done in the same way as for all other bodies in MuJoCo, i.e. from `mjData`, for example using the variable `qpos`. See the colab tutorial for more information.

closed 2023-01-03 by quagla

makitayugo, 2023-01-30:
Using ```print(physics.named.data.geom_xpos)```, we could get the coordinates of the object, but the coordinates of the initial point will always be displayed, even if the object is manipulated and its position changes. Is there any way to display the coordinates accordingly when the object is moved?