google-deepmind/mujoco #679: Get contact force in global reference frame. Collisions with world body.

v-r-a, 2022-12-29:
Hello,

I'm a student using MuJoCo (2.3.1) in C/C++.

I am having trouble getting the correct orientation of contact forces in the global frame.

Here are images illustrating my setup:
This is a snap of a biped robot with a floating base in contact with the ground.
[image]

[image]

Here is the code snippet generating the shown outputs:
```
        mj_contactForce(mm, dd, i, grf_loc);                          // Output in contact frame
        mju_mulMatTVec(grf_glo, dd->contact[i].frame, grf_loc, 3, 3); // Get fx fy fz in the global frame
        printf("con frame x:(%.4f,%.4f,%.4f)\n", dd->contact[i].frame[0], dd->contact[i].frame[1], dd->contact[i].frame[2]);
        printf("    grf_loc:(%.4f,%.4f,%.4f)\n", grf_loc[0], grf_loc[1], grf_loc[2]);
        printf("    grf_glo:(%.4f,%.4f,%.4f)\n", grf_glo[0], grf_glo[1], grf_glo[2]);
```
`condim = 3`
Contact frame and contact force visualization are set through `mjFRAME_SITE` and `mjFRAME_CONTACT` flags, respectively.
The yellow boxes are part of the world body.
The world body has three geometries- a plane and two box primitives.
The foot link body has five geometries- mesh and four primitive spheres at the corners.

You can see in the image and terminal output that the contact frame orientations have flipped when different bodies are colliding with the world body (different geometries, though).

Question 1
I need to obtain ground reaction forces in the global frame.  I am fine with getting either force acting on the world body or by the world body (as shown by arrows).

I am getting some of them as I want and others in a flipped orientation. See the printed o/p `grf_glo`.

How do I get them correctly in a unified manner? 

Question 2
The contact frame given by d->contact[i].frame belongs with which body? One with a higher id or a lower one?  Can you expand on [contact sorting](https://mujoco.readthedocs.io/en/latest/programming/simulation.html?highlight=local%20to%20global%20frame#contacts)?

Question 3
The contact frame normal is along the local x-axis. Shouldn't the frame depiction show the same? (Red cylinder as contact normal instead of blue one).

Thank you

saran-t, 2022-12-29:
For questions 1 and 2, you should check `con.geom1` and `con.geom2` to determine the direction of the contact normal.

For question 3, the x-axis belongs to the contact frame (by definition the contact normal always points along the x-axis of the contact frame), but you're visualising the site frame (which appears in this case to be aligned with the global frame).

v-r-a, 2022-12-29:
Thank your prompt clarifications.

For question 3,
Pardon me. I wrote it wrong here.
Contact frame and contact force visualization are set through mjFRAME_CONTACT and mjVIS_CONTACTFORCE, respectively.  I guess still it is some site frame that is shown and not the contact frame.
[image]

For question 1, I guess the normal points away from `con.geom1`.

For question 2, I guess the frame belongs to the body of `con.geom1` except when it is the world body. (Reason I am pursuing this is to retrieve the direction of friction forces).

yuvaltassa, 2022-12-30:
Regarding frame orientations (question 3), that is a good point. For reference, this happens in [this line](https://github.com/deepmind/mujoco/blob/0d52feaa948b92f4ff0e2f844cb51c1d38df1b79/src/engine/engine_vis_visualize.c#L110).

My rough explanation for this decision is as follows. The internal structure of `contact->frame` is not really a rotation matrix. For one, its axes are on the rows, not the columns (unlike all the other `xmat`s), so if anything it's a **transposed** rotation matrix. Secondly, the semantics of this frame are [normal, trangent1, tangent2], not [x, y, z], so in any case we need to do some operation to convert this frame into a "standard" frame. In many other places in MuJoCo the **z** axis is used to mean the "special" axis:
- z is the default direction of gravity.
- Cameras look down the z-axis.
- The main axis of capsules and cylinders is z, in the geom frame.

So in MuJoCo if one axis of the 3 is "special", it's the z-axis. This is why we decided to use this visualization for the contact frames. Does that makes sense?

closed 2022-12-30 by yuvaltassa

v-r-a, 2022-12-30:
Initially I thought maybe the line defining axes could have been written after transposing `mat`. I was not sure if it was a careful decision.

It's clear with this explanation.

Thanks a lot @saran-t and  @yuvaltassa.