google-deepmind/mujoco #680: I want to which object is in contact with which object using mjContact, etc.

miyukin73, 2022-12-29:
Hello, 

I want to know which object is in contact with which object using mjContact, etc.

If there is contact between objects as shown in the figure below (yellow area), 
is it possible to know which object is in contact with which object, 
such as **geom name**, **id number,** etc. using `mjContact`, etc.?

[image]
[image]

Thank you!

v-r-a, 2022-12-29:
See if you find [this](https://github.com/deepmind/mujoco/issues/132#issuecomment-1229432247) useful.

miyukin73, 2022-12-30:
Hi, @v-r-a 

Thanks to your reply, I was able to solve this problem!

But I would like to ask one more question, if possible, and I am very sorry that it is a basic question.

I would like to obtain the contact force applied to an object.
To do this, I used `cfrc_ext` and others, but the value was always 0, even though the objects are in contact with each other.

On the other hand, using `qfrc_smooth`, `qacc_smooth` and `qfrc_constraint`, I was able to obtain non-zero values.

What is the difference between these two?

Also, are we getting the correct contact forces between objects using `qfrc_smooth`, `qacc_smooth` and `qfrc_constraint` ?

Thank you very much for your time, and we would greatly appreciate your response.
Thank you in advance.

yuvaltassa, 2022-12-30:
You want [mjContactForce](https://mujoco.readthedocs.io/en/latest/APIreference.html#mj-contactforce)

closed 2022-12-30 by yuvaltassa