google-deepmind/mujoco #689: Simple height field produces unexpected collisions/penetrations with the surface

itamar-dw, 2023-01-10:
Hi,
I am experimenting with using height field file, the final goal is simulating a car going on a surface which is not totally flat
I read the relevant part in the documentation, about the limited number of contact points which is taken into account for collisions with a height field. However, the problem I ask about exists even for a flat height field, which I now describe.

I am testing with the most simple height field: 4-points of height `1.0`, which I expect should behave as a `plane` geom element (perpendicular to gravity). This height field can be created in Python by the following code:
```python
import struct
dim = [2, 2]
data = [1., 1., 1., 1.]
with open('terrain.bin','wb') as fout:
    fout.write(struct.pack('2i', *dim) + struct.pack('4f', *data))
```

The problem is that when I add a small sphere and let if fall or drag it around in the "Simulate" app, the sphere behaves like it underwent a collision: it jump around erratically or just fly away. If I pause the simulation when the sphere hits the ground and I look at the `info` section in the app, sometimes (depending on sphere size) the "Size" line reads `Size 200 (50 con)` which I guess means 50 contacts. This is different from using a `plane` geom where the number of contacts is 1. 
So my question is, is it all the result of penetration due to discarded contacts? If yes, why does it happen for a 2X2 height field? I expect it to behave just like a `plane` geom.

Here is a model which explains my question:
```XML
<mujoco>
<asset>
    <!--- create 'terrain.bin' in python using:
        import struct
        dim = [2, 2]
        data = [1., 1., 1., 1.]
        with open('terrain2X2.bin','wb') as fout:
            fout.write(struct.pack('2i', *dim) + struct.pack('4f', *data))
    -->
    <hfield name="terrain" file="terrain2X2.bin" size="5 5 1 .5"/>
</asset>

<asset>
    <texture name="grid" type="2d" builtin="checker" width="512" height="512" rgb2="0 0 0" rgb1="1 1 1"/>
    <material name="grid" texture="grid" texrepeat="2 2" texuniform="true" reflectance=".6"/>
</asset>

<worldbody>
    <geom name="ground" type="hfield" hfield="terrain" size="5 5 .1" pos="0 0 -.5" material="grid"/>
    <body name="sphere">
      <freejoint/>
      <geom size=".2" pos="0 0.0 3" rgba=".2 .8 .2 1"/> 
    </body>
</worldbody>
</mujoco>
```

yuvaltassa, 2023-01-10:
As mentioned in the [documentation](https://mujoco.readthedocs.io/en/latest/XMLreference.html#asset-hfield): 
> To avoid penetration due to discarded contacts, the spatial features of the height field should be large compared to the geoms it collides with.

itamar-dw, 2023-01-11:
Thanks, I of course read that part in the documentation. A flat height field has no spatial features, right? Or am I missing the exact meaning of "spatial features"
I expect that a flat height field (defined by only 4 points of the same height) would behave like a "plane" geom. Is this true?

yuvaltassa, 2023-01-11:
"spatial features" are the triangles (prisms) that make up the HF. If you object can touch more than 2-3 triangles at a time, your HF is too small.

closed 2023-04-30 by kevinzakka