google-deepmind/mujoco #690: How to Optimize or validate the parameters for Damping,Amature, and Stiffness?

DanielCastillo03, 2023-01-10:
**How to ask for help**

Hi,

I'm a student and I'm trying to use MuJoCo.

I have a model that was converted by:
[](https://github.com/aikkala/O2MConverter) and it seems do be unstable during the training that I run. How can validate or optimize the parameters above to maybe help with the instability?

Here is the model:
[Rajagopal2015_converted.zip](https://github.com/deepmind/mujoco/files/10385200/Rajagopal2015_converted.zip)

dulucas, 2023-01-12:
same question here

yuvaltassa, 2023-01-13:
What you are asking for is a system identification package. This is
something we will be working on during 2023.

In the meantime I would recommend hand-tuning the parameters until your
system behaves as expected, more-or-less. The most important parameters for
stability are damping and armature.

Try to create a tight loop where you change a parameter and then examine
the effect in an interactive simulation. Note that the simulate viewer can
inject smooth noise into actuators, which is often very useful in such
cases.

dulucas, 2023-01-13:
Thanks for the reply! It was actually what I did to create the new model😂 Looking forward for the updates of the system identification package!

closed 2023-01-16 by kevinzakka

amdee, 2024-01-19:
@yuvaltassa just a follow-up question, I am interested in leveraging MuJoCo's capabilities for system identification, specifically to identify simulator parameters like mass. My goal is to use a gradient optimization method, utilizing data collected from a physical robot such as the Unitree A1 robot. I want to inquire whether, with the current state of MuJoCo, it is possible to perform system identification tasks of this nature. Are there features or functionalities within MuJoCo that facilitate the optimization-based identification of simulator parameters using data obtained from a real-world robot, and if so, could you provide insights or guidance on how to approach such a task?

Thank you in advance,
Abdul.