google-deepmind/mujoco #692: Can't open viewer with the command"python -m mujoco.viewer" ; Error info:libGL error: MESA-LOADER...

cidxb, 2023-01-11:
Hi,

I'm looking for some help with  the GUI of Mujoco 

**System Info** 
Ubuntu 22.04 
NVIDIA driver 525
Kernel 6.0.0-1010-oem #10-Ubuntu

**Error message** 
libGL error: MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
/home/jeffery/.local/lib/python3.10/site-packages/glfw/__init__.py:912: GLFWError: (65543) b'GLX: Failed to create context: BadValue (integer parameter out of range for operation)'
  warnings.warn(message, GLFWError)
ERROR: could not create window

Press Enter to exit ...

**Additional**

I successfully opened it at the beginning.
But then I have accidentally run "apt upgrade"
Then occur this problem, the Graphic driver problem(nvidia-smi not showing correctly).

But I have already used the --purge command to remove all graphic driver and reinstall it.
The NVIDIA display driver is functional now
But this Problem with the Mujoco viewer still.

Please anyone know how to solve it?

I have read some relevant posts online regarding libGL error, however, they were not helpful.
Thank you !

cidxb, 2023-01-13:
The problem is solved !!
It is because the apt update makes the soft links of lib break.
use ll command to check the file 

strings libstdc++.so.6 | grep GLIBCXX_3.4.30  (if broke can't find it )

locate libstdc++.so.6  (locate workable file ) 

select one (should be the /usr/lib/x86_64-linux-gnu one )
using the str command above  to grep again, the fine is here!

rm the problematic file in  dictionary 

ln -s the dictionary of good file to the problematic dichotomy .

 (create a softlink to the correct location )

Anyway, it its roughly the solution of my case

closed 2023-01-13 by cidxb

coder-rakeshpaul, 2024-04-15:
@cidxb could you please explain this fix in simple terms? I am currently facing the same issues. I am also on ubuntu 22.04

cidxb, 2024-04-16:
> @cidxb could you please explain this fix in simple terms? I am currently facing the same issues. I am also on Ubuntu 22.04
It is basically a broken file soft link problem, which means the file you want is not here really, but an indicator of where it is, but this link broke, and then the program is unable to find it. SO WE WANT TO REBUILD THE LINK.

1. use `ll `command to check the file(in this case, `ll iris_dri.so`)  ###Check the file link
2. use command `strings libstdc++.so.6 | grep GLIBCXX_3.4.30`   ###find this missing .so library
3. use the command `locate libstdc++.so.6`  ###To find where in your system has this file/libray
4. remove the old/problematic file which has been reported as _`No such file or directory_ 
5. use the command`  ln -s`  to link the dictionary of the good file to the problematic dictionary.  ###create a soft link to the correct location )

cidxb, 2024-04-17:
> @cidxb I am sorry I am extremely new to linux so this might sound a dumb question I followed your instructions up till 3, and getting this when I try the `locate libstdc++.so.6` ![image](https://private-user-images.githubusercontent.com/114281050/322939838-87cfad6e-a08a-4fe5-8d19-f387fd38f1c7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMzNjEzMTksIm5iZiI6MTcxMzM2MTAxOSwicGF0aCI6Ii8xMTQyODEwNTAvMzIyOTM5ODM4LTg3Y2ZhZDZlLWEwOGEtNGZlNS04ZDE5LWYzODdmZDM4ZjFjNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxN1QxMzM2NTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NzJkZjRmNDIyY2I4MDEyZTg3NzUwZmE2MjA2MDNhY2ExODNlMzdhN2Y4ZTg4N2VlNjViNDM3YzVlN2MxNGQ1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.3GJD0KSkjKYxowG_1J_GEqpyuF2ELz_vQpbVno8qViQ)
> 
> I don't understand step 4 and 5, how am I supposed to know which is the problematic file? (i would have tried the same file but I don't see that in the options given)

What is your original error trance back? You can check it.  That should state what file went wrong, you delete it, and follow the step 4 and 5 to create a new file that can work.  It is very simple.

coder-rakeshpaul, 2024-04-17:
@cidxb 
this is the snippet of the error

[image]

So from this I am inferring that I should use `rm usr/lib/dri/iris_dri.so `

Then in the 5th step I should use `ln -s` to the correct location, but which would be the correct location?
this is the output I get when I try `locate libstdc++.so.6`
[image]

Is it this `/usr/lib/x86_64-linux-gnu/libstdc++.so.6` ?
(I am sorry I'm doing low effort stuff here by not running this stuff and checking it, cause I have some important changes saved and re-configuring ubuntu again would be a pain (I know this cause I screwed up once before and was set back 3 days behind))

CRLqinliang, 2024-05-02:
> 感谢大家的帮助，我终于弄清楚了。 如果有人还有其他疑问，可以查看此[stackoverflow 帖子](https://stackoverflow.com/questions/72110384/libgl-error-mesa-loader-failed-to-open-iris)

Solve my problem. Thx

kingchou007, 2025-03-12:
conda install -c conda-forge libstdcxx-ng

I tried this, fix issue. Hoe this works for you guys