google-deepmind/mujoco #698: I can't grab it after changing to ``condim="3"``

makitayugo, 2023-01-17:
I can't grab it after changing to ``condim="3"``
We believe that the arm is not gripping enough.
If there is a way to do this, we would like to know.
This xml file is based on ``scene.xml``, shown below.
https://github.com/deepmind/mujoco_menagerie/blob/main/franka_emika_panda/scene.xml

_Originally posted by @makitayugo in https://github.com/deepmind/mujoco/issues/677#issuecomment-1379843003_

Balint-H, 2023-01-17:
In cases like this enabling [`multiccd` flag])(https://mujoco.readthedocs.io/en/stable/XMLreference.html#option-flag) is often recommended, might be worth trying it. If that doesn't work you could also try disabling slipping with the [noslip option](https://mujoco.readthedocs.io/en/stable/XMLreference.html#option), however, that may not be desirable.

makitayugo, 2023-01-31:
Thank you very much. I tried ```multiccd```, but as before, the grab doesn't last long and the robotic arm drops things quickly.

erez-tom, 2023-01-31:
You should try adding "noslip_iterations=2" (or higher, if necessary): it
was designed exactly to remedy this mode of failure.

makitayugo, 2023-01-31:
I have added ```<option noslip_iterations="10"/>``` but do not see any change.
Is there a mistake in the code?
```XML<mujoco model="panda scene">

  <include file="panda.xml"/>

  <statistic center="0.3 0 0.4" extent="1"/>
  <extension>
    <required plugin="mujoco.elasticity.cable"/>
  </extension>

  <visual>
    <headlight diffuse="0.6 0.6 0.6"  ambient="0.3 0.3 0.3" specular="0 0 0"/>
    <rgba haze="0.15 0.25 0.35 1"/>
    <global elevation="-30"/>
  </visual>
  <size nconmax="5000" njmax="10000"/>
  
  <option noslip_iterations="10"/>
      

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512"
        height="3072"/>
    <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4"
        rgb2="0.1 0.2 0.3" markrgb="0.8 0.8 0.8" width="300" height="300"/>
    <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5"
        reflectance="0.2"/>
  </asset>

  <worldbody>
    <light pos="0 0 1.5" dir="0 0 -1" directional="true"/>
    <geom name="floor" size="0 0 0.05" type="plane" material="groundplane"/>
    <composite type="cable" curve="s" count="41 1 1" size="1.5" offset="-1 0.3 .01"  initial="none">
      <plugin plugin="mujoco.elasticity.cable">
        <!--Units are in Pa (SI)-->
        <config key="twist" value="1e3"/>
        <config key="bend" value="1e4"/>
      </plugin>
      <joint kind="main" damping=".001"/>
      <geom type="capsule" size=".004" rgba=".8 .2 .1 1"  friction=".1" condim="3"/>
    </composite>
    <body name="adapter" pos=".5 0.3 .01">
      <geom type ="sphere" size=".0055 "  friction=".1" condim="3"/>
      <freejoint/>
    </body>
    <body name="box" pos=".5 0 .1">
      <geom type ="box" size=".007 .007 .007 " friction=".1" condim="3"/>
      <freejoint/>
    </body>
    <!--<body name="socket" pos="1 0.3 .1">
      <geom type="box" size="0.05 0.05 0.05" margin=".02" gap=".01" rgba=".1 .8 .2 1" contype="2" conaffinity="2"/>
    </body>-->
    <!--<body name="box1" pos="0.1 0.3 .1" zaxis="0 0 1">
      <geom type="box" size="0.05 0.05 0.05"  group="1" condim="3"/>
    </body>
    <body name="box2" pos="0.5 0.3 .1" zaxis="0 0 1">
      <geom type="box" size="0.05 0.05 0.05"  group="1" friction=".1" condim="3"/>
    </body> -->
  </worldbody>
  <equality>
    <connect name="right_boundary" body1="B_last" body2="adapter" anchor=".0376 0 0"/>
  </equality>
  <contact>
    <exclude body1="B_last" body2="adapter"/>
  </contact>>

</mujoco>
```
[Screencast from 2023年01月31日 20時09分18秒.webm](https://user-images.githubusercontent.com/85473041/215744440-f3e63e7f-3984-429a-863d-79bef428e47a.webm)

yuvaltassa, 2023-02-08:
Hi,

Please can you repeat your test under the following conditions. 

0. Before anything, comment out the cable so you have real-time simulation, and reduce `noslip_iterations` to 3. More than 3 will just slow you down and have no other effect.

Now repeat your test, making these modifications (separately, not at the same time):

1. Try `<option timestep="x"/>` with x 1e-4 or 1e-5. What happens?
2. Reduce the gripper gains. What happens?
3. Use a larger object. What happens?

I want to rule out some basic possibilities:
- The timestep is too large for the velocities and object size (i.e. in 1 timestep the object is moving too much relative to it's own size, leading to penetrations)
- The grip force is unreasonably large and the object "pops out" due to instabilities.

kevinzakka, 2023-11-22:
Assuming this has been resolved. Feel free to re-open if not.

closed 2023-11-22 by kevinzakka