google-deepmind/mujoco #700: Python renderer class using egl and osmesa

heleiduan, 2023-01-19:
The following errors show up when using `egl` and `osmesa` for rendering via Python bindings. 

Minimal example is the same as tutorial as below, 

```
import os
os.environ['MUJOCO_GL']='egl'

import mujoco
import numpy as np
import matplotlib.pyplot as plt

if "MUJOCO_GL" in os.environ:
    print(os.getenv('MUJOCO_GL'))
    print(os.getenv('PYOPENGL_PLATFORM'))

xml = """
<mujoco>
  <worldbody>
    <light name="top" pos="0 0 1"/>
    <body name="box_and_sphere" euler="0 0 -30">
      <joint name="swing" type="hinge" axis="1 -1 0" pos="-.2 -.2 -.2"/>
      <geom name="red_box" type="box" size=".2 .2 .2" rgba="1 0 0 1"/>
      <geom name="green_sphere" pos=".2 .2 .2" size=".1" rgba="0 1 0 1"/>
    </body>
  </worldbody>
</mujoco>
"""

model = mujoco.MjModel.from_xml_string(xml)
renderer = mujoco.Renderer(model)
data = mujoco.MjData(model)

renderer.enable_depth_rendering()

mujoco.mj_forward(model, data)
renderer.update_scene(data)

depth = renderer.render()
depth -= depth.min()
depth /= 2 * depth[depth <= 1].mean()
img = 255 * np.clip(depth, 0, 1)
plt.imshow(img.astype(np.uint8))
plt.colorbar(label='Distance to Camera')
plt.show()
```

For `EGL` option, rendering is successful but the `__del__` failed, 

> Exception ignored in: <function GLContext.__del__ at 0x7fa98b52ee50>
  Traceback (most recent call last):
    File "/home/anaconda3/envs/env1/lib/python3.8/site-packages/mujoco/egl/__init__.py", line 130, in __del__
    File "/home/anaconda3/envs/env1/lib/python3.8/site-packages/mujoco/egl/__init__.py", line 123, in free
    File "/home/anaconda3/envs/env1/lib/python3.8/site-packages/OpenGL/error.py", line 230, in glCheckError
  OpenGL.raw.EGL._errors.EGLError: <exception str() failed>

For `osmesa`, 
> Traceback (most recent call last):
  File "test.py", line 4, in <module>
    import mujoco
  File "/home/anaconda3/envs/env1/lib/python3.8/site-packages/mujoco/__init__.py", line 46, in <module>
    from mujoco.gl_context import *
  File "/home/anaconda3/envs/env1/lib/python3.8/site-packages/mujoco/gl_context.py", line 38, in <module>
    from mujoco.osmesa import GLContext as _GLContext
  File "/home/anaconda3/envs/env1/lib/python3.8/site-packages/mujoco/osmesa/__init__.py", line 31, in <module>
    from OpenGL import GL
  File "/home/anaconda3/envs/env1/lib/python3.8/site-packages/OpenGL/GL/__init__.py", line 4, in <module>
    from OpenGL.GL.VERSION.GL_1_1 import *
  File "/home/anaconda3/envs/env1/lib/python3.8/site-packages/OpenGL/GL/VERSION/GL_1_1.py", line 14, in <module>
    from OpenGL.raw.GL.VERSION.GL_1_1 import *
  File "/home/anaconda3/envs/env1/lib/python3.8/site-packages/OpenGL/raw/GL/VERSION/GL_1_1.py", line 7, in <module>
    from OpenGL.raw.GL import _errors
  File "/home/anaconda3/envs/env1/lib/python3.8/site-packages/OpenGL/raw/GL/_errors.py", line 4, in <module>
    _error_checker = _ErrorChecker( _p, _p.GL.glGetError )
AttributeError: 'NoneType' object has no attribute 'glGetError'

 
- Include the following context:
  - Operating system.
 ```
    Ubuntu 22.04.1 LTS
    NVIDIA-SMI 470.161.03   Driver Version: 470.161.03   CUDA Version: 11.4
```
  - MuJoCo version (and if the bug is new, the version where it used to work).
    -  2.3.1.post1
  - For Python issues, what bindings are you using (i.e `mujoco`, `dm_control`, `mujoco-py`)?
    - `mujoco`

saran-t, 2023-02-06:
Looking at your stack trace this doesn't seem to be MuJoCo-specific? Both errors seem to originate from PyOpenGL...

Cpupshis, 2023-02-24:
I have the same problem for egl. 

> Looking at your stack trace this doesn't seem to be MuJoCo-specific? Both errors seem to originate from PyOpenGL...

Tried reinstalling both OpenGL and MuJoCo and updating drivers but no luck

sebimarkgraf, 2023-03-05:
I can confirm the suspicion of @saran-t .
The problem seems to be a missing symlink that pyopengl assumes.

A workaround can be found in https://github.com/mcfletch/pyopengl/issues/90 and a fix is implemented in https://github.com/mcfletch/pyopengl/pull/93 but not yet merged.
I suppose this can be closed here, as this does not seem to be an issue of mujoco itself.

heleiduan, 2023-03-05:
@sebimarkgraf Thanks for pointing this out!

cidxb, 2023-03-13:
> I can confirm the suspicion of @saran-t . The problem seems to be a missing symlink that pyopengl assumes.
> 
> A workaround can be found in [mcfletch/pyopengl#90](https://github.com/mcfletch/pyopengl/issues/90) and a fix is implemented in [mcfletch/pyopengl#93](https://github.com/mcfletch/pyopengl/pull/93) but not yet merged. I suppose this can be closed here, as this does not seem to be an issue of mujoco itself.

Tried it, but not working for me.

CHbloom, 2023-04-02:
have you solve it?I met the same problem

sebimarkgraf, 2023-04-02:
@CHbloom are you having the same stacktrace as the OP?
In that case you could verify if the same link is missing in your files.

If thatvis the case the hack should work.
Otherwise, the problem should probably raised with PyOpenGL as they have better insights and could assist you in your issues.

closed 2023-04-30 by kevinzakka

traversaro, 2024-05-20:
Apparently this issue appears quite easily in `glCheckError` and `glGetError` related errors for pyopengl, so let's post an explanation message here.

pyopengl does a lot of lazy loading of OpenGL-related shared libraries (i.e. `.so`/`.dylib`/`.dll` files, depending on the operating system). However, it does not raise directly an error if a given library is not found, but suppress the exception generated by the error, see for example https://github.com/search?q=repo%3Amcfletch%2Fpyopengl%20loadLibrary&type=code . So, the error can be generated by:
* Missing symlink of a library that is searched with a given name, and is present in the operating system with another name
* Library missing completely (for example because the corresponding package is not installed)

In general, even if the error seems similar, it is possible that the root cause (i.e. the libraries missing) are completely different.