google-deepmind/mujoco #701: trajectory optimization using mj_inverse / mj_forward

chrishunt93, 2023-01-20:
Hi,

Apologies in advance for the long post.

I'm a PhD student and I'm trying to use MuJoCo for trajectory optimization.

I am accessing MuJoCo via the Python bindings and am having some difficulty calculating an initial guess of the optimal controls needed to seed my optimizer. Here are the steps I've used to try and solve this problem.

1. I define an initial guess for joint position (q0) that is a straight line in configuration space with a bell-shaped velocity profile.
2. I compute the initial guess for velocity (dq0) as the gradient of q0 divided by dt.
3. I compute the initial guess for acceleration (ddq0) as the gradient of dq0 divided by dt
4. I then compute the initial guess for the generalized forces (tau0) using the mj_inverse function (I suspect I am setting up this part incorrectly). Here is the relevant snippet of the code:
```
# initial guess for kinematics
q0 = np.array( [ qi + ( qf - q0 ) * ( ( 10.0 * ( t / tf ) ** 3 ) 
                                    - ( 15.0 * ( t / tf ) ** 4 ) 
                                    + (  6.0 * ( t / tf ) ** 5 ) ) for t in ts ] )
dq0 =  np.gradient( q0, axis = 0 ) / dt
ddq0 = np.gradient( dq0, axis = 0 ) / dt

# initial guess for dynamics
tau0 = []
for i in range( len( ts ) ):
    data.qpos = q0[i,:]
    data.qvel = dq0[i,:]

    data.qfrc_applied = np.zeros( shape = ( model.nq, ) )
    data.ctrl = np.zeros( shape = ( model.nu, ) )

    mujoco.mj_forward( model, data ) # computes mass matrix and bias forces
    data.qacc = ddq0[i,:]            # set the generalized accelerations
    mujoco.mj_inverse( model, data ) # compute the generalized forces via inverse dynamics

    tau0.append( data.qfrc_inverse.copy() )
tau0 = np.vstack( tau0 )
```
5.  Finally, I compute the initial guess for the change in generalized forces ( dtau0, which is the input to my optimizer) as the gradient of tau0 divided by dt.

**The problem I am running into is that when I provide dtau0 as input to a dynamics simulation, I do not get back q0. In fact, for small accelerations (~1e-2), the inverse dynamics equation returns extremely large generalized forces (~1e5). This seems intuitively incorrect somehow.** 

This suggests to me that I am not setting up the inverse or forward dynamics function calls correctly (perhaps I'm not zeroing out some other forces?). Here is a snippet of the code I use to run that simulation.

```
def update( x, u ):
    # x is a combined state vector [q, dq, tau]    
    q = x[:model.nq].copy()
    dq = x[model.nq:(2*model.nq)].copy()
    tau = x[-model.nq:].copy()

    # u is the input control vector [dtau]
    dtau = u.copy()

    # use MuJoCo dynamics to get joint accelerations
    data.qpos = q
    data.qvel = dq
    data.qfrc_applied = tau

    mujoco.mj_forward( model, data )
    ddq = data.qacc.copy()

    # return the state update
    return np.hstack( [dq, ddq, dtau] )
```
```
def simulation(x0, U):
    X = [x0]
    for i in range( U.shape[0] - 1 ):
        dx = update( X[-1], U[i,:] )
        X.append( X[-1] + dx * dt )
    return np.vstack( X )
```

Any insight you could give to this setup would be appreciated. At its core, I think this is some misunderstanding I have about the mj_inverse and mj_forward functions.

I've attached a zip of the model I am using for this project if it helps. It is originally an OpenSim model that was automatically converted using the O2MConverter tool.

Just a note, I've changed a few flags of the MuJoCo simulation from default values. Here is the code snippet I used to do that:
```
model.opt.disableflags = 1 << 4 # disable contact constraints
model.opt.integrator = 1 # change integrator to RK4
```

[mjcf.zip](https://github.com/deepmind/mujoco/files/10464041/mjcf.zip)

yuvaltassa, 2023-01-20:
I have not looked at your model however:
1. The large forces suggest some constraints are still active. Can you please disable the entire constraint solver?
2. If you want the forward and inverse dynamics to match, you cannot use the RK4 integrator. Apologies that this is not documented. In fact it might be the case that only the Euler integrator is supported since I don't think we tested this for the implicit integrator which was added quite recently.
3. There is an enable flag to test that the forward and inverse dynamics match, which should be helpful.

Do some more investigations using the above tips and let's continue the conversation here. Generally speaking, you seem to be doing the right thing (again, without looking very carefully).

chrishunt93, 2023-01-20:
Thank you for your reply. After some investigation, here is what I've found:

1. I believe you are correct. When all constraints are inactive, the computed generalized forces (qfrc_inverse) are much more reasonable (and result in the better coherence between mj_inverse and mj_forward). More specifically, it seems that the contact constraints (which were initially disabled) and the **equality constraints** are the biggest contributors.
2. I've switched back to the Euler integrator. Thank you letting me know.
3. I've tried using the enable flag and constantly receive a value of [0,0] when I probe data.solver_fwdinv. Unfortunately, I'm not exactly sure how I should interpret this.

As it stands, the forward simulation given dtau0 has good agreement with the expected kinematics when I disable both contact and equality constraints. Unfortunately, viewing the simulation with equality constraints disabled results in some unrealistic relationships between model bodies (and so I'd prefer these constraints be enabled).

Is there a way via the python API to access the constraint forces expressed in generalized coordinates? Perhaps then I could plot them over the course of a trajectory.

Additionally, I'm unsure if the equality constraints have been defined reasonably in the model (as it was converted originally from OpenSim). Upon visual inspection in the MuJoCo simulator, it seems fine, however, I could definitely be wrong here. I would appreciate any guidance on how to make this more stable / well-behaved.

Thank you for your time.

yuvaltassa, 2023-02-01:
`data.solver_fwdinv` values of [0, 0] is what you want, it means the forward and inverse dynamics are a perfect match.

In Python you can do `data.qfrc_constraint` (or any other member of `mjData`)

closed 2023-02-01 by yuvaltassa