google-deepmind/mujoco #703: Analytical Derivative of Inverse Dynamics wrt Acceleration

Daniellayeghi, 2023-01-20:
Hello, 

I have written my own finite difference implementation for this derivative aiming to reformulate something similar to the [CIO](https://homes.cs.washington.edu/~todorov/papers/MordatchSIGGRAPH12.pdf) paper. In the docs however you define this derivative as: 

$\frac{d \tau}{d \dot{v}}=M +  J^{T} + H[s] + J$

which I assume can be computed analytically since $s$ is well defined. Is this assumption correct and if so is there a simple way to access it? 

Many thanks.

yuvaltassa, 2023-01-24:
Great question. I have bad news and good news 🙂 

1. Bad news: The full $H$ of the primal problem described in the [documentation](https://mujoco.readthedocs.io/en/latest/computation.html#reduced-primal-problem) is unrelated to the `contact.H` matrix. They are both just H for Hessian. The full $H$ is indeed analytically computable, but we have not gotten around to writing the analytical derivative.
2. Good news: The inverse dynamics are blazingly fast, see figure 3 [here](https://homes.cs.washington.edu/~todorov/papers/TodorovICRA14.pdf), so a finite-difference function would be fine and in fact often better since it will automatically finite-difference user code, as we do in the [MJPC](https://github.com/deepmind/mujoco_mpc) project. This is another thing we have not gotten around to, but is actually quite easy. If you follow the pattern of [`mjd_stepFD`](https://github.com/deepmind/mujoco/blob/cd99fbd59dc75fa8be97fe90a391d2055ee0a4d6/src/engine/engine_derivative.c#L1479) you could easily write an `mjd_inverseFD` that gives you all possible derivatives.

Let us know if you might be interested in writing that, it would be a very useful contribution.

closed 2023-01-24 by yuvaltassa

Daniellayeghi, 2023-01-26:
> Great question. I have bad news and good news slightly_smiling_face
> 
>     1. Bad news: The full H of the primal problem described in the [documentation](https://mujoco.readthedocs.io/en/latest/computation.html#reduced-primal-problem) is unrelated to the `contact.H` matrix. They are both just H for Hessian. The full H is indeed analytically computable, but we have not gotten around to writing the analytical derivative.
> 
>     2. Good news: The inverse dynamics are blazingly fast, see figure 3 [here](https://homes.cs.washington.edu/~todorov/papers/TodorovICRA14.pdf), so a finite-difference function would be fine and in fact often better since it will automatically finite-difference user code, as we do in the [MJPC](https://github.com/deepmind/mujoco_mpc) project. This is another thing we have not gotten around to, but is actually quite easy. If you follow the pattern of [`mjd_stepFD`](https://github.com/deepmind/mujoco/blob/cd99fbd59dc75fa8be97fe90a391d2055ee0a4d6/src/engine/engine_derivative.c#L1479) you could easily write an `mjd_inverseFD` that gives you all possible derivatives.
> 
> 
> Let us know if you might be interested in writing that, it would be a very useful contribution.

Thanks a lot for the reply. I guess in that case FD might even be more efficient if $H$ is not freely available. I'm very happy to contribute. I think I am pretty familiar with the [`mjd_stepFD`](https://github.com/deepmind/mujoco/blob/cd99fbd59dc75fa8be97fe90a391d2055ee0a4d6/src/engine/engine_derivative.c#L1479) structure. That being said should I just implement and test it following the contribution guidelines or are there things that need to be coordinated?

yuvaltassa, 2023-01-26:
Your willingness to contribute is appreciated!

Following the [contribution guidelines](https://github.com/deepmind/mujoco/blob/main/CONTRIBUTING.md) should be sufficient, though note the style of the C files is [somewhat idiosyncratic](https://github.com/deepmind/mujoco/blob/main/STYLEGUIDE.md), but maintaining self-consistency should be straightforward. We are pretty strict about it so I fon't think you'll encounter any style ambiguities when looking at the existing codebase.

The only real thing that needs to be thought about is whether this needs to be a completely separate function or if `mj_stepFD` could be augmented with a flag that will switch it from forward to inverse. The benefit would be that it would save some code duplication, but I think the downside would be greater, since in the inverse case the dimensions are different (`qacc` is `nv` while `ctrl` is `nu`) and a flag that changes the expected dimensionality of the outputs sounds like asking for trouble. So I think having some code duplication is a reasonable price to pay in this case. @nimrod-gileadi, would you agree?

yuvaltassa, 2023-01-26:
I should add that, in case you missed it, the tests for these functions live [here](https://github.com/deepmind/mujoco/blob/main/test/engine/engine_derivative_test.cc).

Daniellayeghi, 2023-01-30:
> Your willingness to contribute is appreciated!
>
> Following the [contribution guidelines](https://github.com/deepmind/mujoco/blob/main/CONTRIBUTING.md) should be sufficient, though note the style of [...]

Thanks for the info! Looking at the code the implicit function of  `mj_stepFD` is defined by two things: 

- Given the name the derivatives are always w.r.t `mj_step` therefore forward dynamics.
- The function signature takes `DyDu` and `DsDu` which is irrelevant to inverse dynamics but technically should be 0? 
- The `next` variables take the dimensionality of  `nq + nv + nu` and the corresponding functions in the FD logic assume this dimensionality e.g. `DyDv` implicitly calls `stateDiff`.

However otherwise most of functionality in the code would be the same but w.r.t a different function i.e. `mj_inverse` where the FD logic doesn't need to compute `stateDiff` but rather something similar to `diff` (not diff itself however since it's arguments are not the same as `stateDiff`) . Can this potential flag simply define a ptr to either `mj_step` or `mj_inverse` and another ptr which points to either `stateDiff` for forward or something similar to `diff` with the same signature as `stateDiff` for inverse . I thought this because from the interface it seems like the issue of the dimensionalty of the output  `DyD`  and `DsD` seems to be offloaded to the user since they are arguments to `mj_stepFD` and as long as it corresponds to the function that is being differentiated the code should work.

Hopefully this is not too convoluted thought I share my impression of the code as a user.

yuvaltassa, 2023-02-04:
The are other differences, for example in the forward dynamics there is the nontrivial handling of clamped `ctrl`, which doesn't exist in the inverse dynamics. There _is_ a flag that we could add to `mjd_stepFD`, but it's not step/inverse, but rather step/forward. In other words, what if the user wants the derivatives w.r.t acceleration rather than the next step. In this case the internal call would be to `mj_forwardSkip` rather than `mj_stepSkip` and the `y` output dimension would be smaller (`nv+na` rather than `2*nv+na`), but everything else would be exactly the same.

So basically, due to the above, I think making an entirely new function for the inverse dynamics would be a better design, rather than trying to shim the inverse dynamics into the existing FD functions.

Does that make sense?