google-deepmind/mujoco #709: Moving sites on key press

aravindsiv, 2023-01-25:
Hi,

I'm a Ph.D. student trying to use MuJoCo for planning among dynamic obstacles.

I'm looking for some help using keyboard events to move sites around.

Here is a model which explains my question:

```XML
<mujoco>
	<option gravity="0 0 -9.81" />
	<worldbody>
        <light diffuse=".5 .5 .5" pos="0 0 3" dir="0 0 -1"/>
		<geom type="plane" size="5 5 0.1" rgba=".9 .9 .9 1"/>
        <site name="human" type="cylinder" pos="2.5 2.5 0.5" size="0.5 1.0" rgba="1 0 0 1"/>
	</worldbody>
</mujoco>
```

Here is the code snippet that I'm using:

```Python
from mujoco.glfw import glfw
import numpy as np

xml_path = 'keyboard.xml' 
simend = 30 

def keyboard(window, key, scancode, act, mods):
    if act == glfw.PRESS and key == glfw.KEY_BACKSPACE:
        mj.mj_resetData(model, data)
        mj.mj_forward(model, data)

    if act == glfw.REPEAT and key == glfw.KEY_RIGHT:
        print("Key pressed!")
        data.site_xpos[0][0] += 0.1

# MuJoCo data structures
model = mj.MjModel.from_xml_path(xml_path)  # MuJoCo model
data = mj.MjData(model)                     # MuJoCo data
cam = mj.MjvCamera()                        # Abstract camera
opt = mj.MjvOption()                        # visualization options

glfw.init()
window = glfw.create_window(1200, 900, "Demo", None, None)
glfw.make_context_current(window)
glfw.swap_interval(1)

mj.mjv_defaultCamera(cam)
mj.mjv_defaultOption(opt)
scene = mj.MjvScene(model, maxgeom=10000)
context = mj.MjrContext(model, mj.mjtFontScale.mjFONTSCALE_150.value)

glfw.set_key_callback(window, keyboard)

cam.azimuth = 90 ; cam.elevation = -45 ; cam.distance =  13
cam.lookat =np.array([ 0.0 , 0.0 , 0.0 ])

while not glfw.window_should_close(window):
    time_prev = data.time

    while (data.time - time_prev < 1.0/60.0):
        mj.mj_step(model, data)

    if (data.time>=simend):
        break

    viewport_width, viewport_height = glfw.get_framebuffer_size(window)
    viewport = mj.MjrRect(0, 0, viewport_width, viewport_height)
    mj.mjv_updateScene(model, data, opt, None, cam,mj.mjtCatBit.mjCAT_ALL.value, scene)
    mj.mjr_render(viewport, scene, context)
    glfw.swap_buffers(window)
    glfw.poll_events()

glfw.terminate()
```

Although the above script registers the key press, it does not update the site's position. Is there a way to make this change persistent?

Thanks!

yuvaltassa, 2023-01-26:
Yes.

You are moving the site in `mjData.site_xpos` which get overwritten at every `mj_step`. `mjData` is all about non persistent stuff. This is explained in the [overview](https://mujoco.readthedocs.io/en/latest/overview.html#key-features). Instead, set `model.site_pos`

closed 2023-01-26 by yuvaltassa

aravindsiv, 2023-01-26:
Thanks, that worked!